gnuton/asuswrt-merlin.ng #487: TUF-AX3000_3004_388.4_0-gnuton1_puresqubi.w - is incredibly unstable

Ichimikichiki, 2023-10-31:
**Router Model Affected**
TUF-AX3000 

**Firmware Version Affected**
3004_388.4_0-gnuton1_puresqubi

**Describe the bug**
Freezes for about 45 seconds out of every 90 seconds

Tried flashing 20 times. Tried flashing back 40 times, but can't get it to stay connected long enough to complete.

gnuton, 2023-10-31:
Was the previous version working fine? I will check it

gnuton, 2023-10-31:
Did you tested the latest beta too? Do you have the same issue?

Ichimikichiki, 2023-11-01:
I did. And to a lesser extent. Only about 10% of the instability.

There are no logged events when this occurs.

gnuton, 2023-11-02:
I am testing the _tuf image and so far so good. 
To test if it freezes I ssh on the router and ping www.google.it

gnuton, 2023-11-03:
Tested a few hours.. not an issue with the _tuf image. I will test the non tuf one.

gnuton, 2023-11-03:
@Ichimikichiki can you please define a test plan I can execute to assess the instability of this image?

Ichimikichiki, 2023-11-06:
> I am testing the _tuf image and so far so good. To test if it freezes I ssh on the router and ping [www.google.it](http://www.google.it)

It even freezes while ssh'd into the router. I have to open a new session.

I'm looking for another router. I'm just gonna toss it.

gnuton, 2023-11-06:
@Ichimikichiki the docker image works as I used it locally and on the cloud. 
The tuf image works to. I have to check the other one. 
The  problem happens on your router. so the question is.. what happens:?
the only thing to understand what's wrong is to check your logs.

gnuton, 2023-11-06:
Get docker images already built with a docker pull. No reason to build your docker image unless you wanna develop the docker image. All docker and tug images are built on the cloud by circleci. If you fork the repo and enable circleci you do not need to setup anything on your local machine. No pain.

Ichimikichiki, 2023-11-09:
So we can close this now.

closed 2023-11-09 by Ichimikichiki