gnuton/asuswrt-merlin.ng #490: TUF-AX3000 missing tabs on tuf fw

alfky12, 2023-11-05:
**Router Model Affected**
TUF-AX3000

**Firmware Version Affected**
TUF-AX3000_3004_388.4_0-gnuton1_tuf_puresqubi

**Is this bug present in upstream Merlin releases too?**
Don't know

**Is this bug present in non-tuf release too?**
Neither I know

**Is this bug present in alpha/beta release too?**
I never try beta, just stable version.

**Describe the bug**
Missing tabs:
1. Traffic classification
2. VPN - Status
That's all I've found so far.
All the missing tabs can still be accessed via a direct link if you/browser still remember them.

**To Reproduce**
Traffic classification tab should be in the Adaptive QoS menu.
VPN - Status tab should be in the VPN menu.

**Expected behavior**
That tabs should be on their respective menus.

I didn't want to try on non-tuf fw because someone reported a very unstable system.
And oh, a little extra, the tools menu doesn't have a wrench icon.
Sorry for bad english, not my native language.

gnuton, 2023-11-06:
this is a regression I will check and fix this.

alfky12, 2023-11-14:
Or maybe, since this is just a webui, can you just make a temporary fix in the form of a tampermonkey userscript to add the missing icon and tabs?

alfky12, 2023-11-14:
Sorry, I clicked the wrong button

gnuton, 2023-11-16:
http://router.asus.com/Advanced_VPNStatus.asp
[image]
and 
http://router.asus.com/QoS_Stats.asp missing
[image]

on TUF are
[image]

and
[image]

gnuton, 2023-11-16:
fixed in f22f512d04

gnuton, 2023-11-17:
test image https://app.circleci.com/pipelines/github/gnuton/asuswrt-merlin.ng/1684/workflows/d545468e-b410-4b15-932d-2e94dccb14b9

closed 2023-11-17 by gnuton