gnuton/asuswrt-merlin.ng #491: [FEATURE REQUEST] VLAN support for ZenWiFi XT8 / RT-AX95Q and ZenWifi ET8 / RT-AXE95Q units

Insan1ty0ne, 2023-11-07:
### **Is your feature request related to a problem? Please describe.**
It is EXTREMELY frustrating that the _slightly_ newer ZenWiFi XT12 Pro / ET12 Pro units have access to a BETA firmware (9.0.0.6.102.4856) from ASUS which provides full VLAN support among many other useful things. Why couldn't this same beta firmware be released for the ZenWiFi XT8 / RT-AX95Q and ZenWifi ET8 / RT-AXE95Q units? They said this firmware was going to release officially "soon" but that was nearly 9-months ago.

### **Describe the solution you'd like**
Add AsusWRT 5.0 (3.0.0.6 Branch) features like VLAN support to the ZenWiFi XT8 / RT-AX95Q and ZenWifi ET8 / RT-AXE95Q units. This could be taken from ASUS's latest BETA firmware version 9.0.0.6.102.4856 for the ZenWiFi XT12 Pro / ET12 Pro units which can be downloaded here: [https://dlcdnets.asus.com/pub/ASUS/wireless/ZenWiFi_ET12/FW_ZenWiFi_ET12_90061024856.zip?model=ZenWiFi%20Pro%20ET12](url)

It could also be taken from ASUS's latest STABLE firmware version 3.0.0.6.102_21514 for the RT-AX88U Pro which can be downloaded here: [https://dlcdnets.asus.com/pub/ASUS/wireless/RT-AX88U%20Pro/FW_RT_AX88U_PRO_300610221514.zip?model=RT-AX88U%20Pro](url)

### **Describe alternatives you've considered**
The following method would "work" but it is what I would consider "messy" and is absolutely NOT an elegant solution in any way: [https://www.snbforums.com/threads/how-to-use-vlans-on-your-non-pro-asus-router-with-386-or-388-code-no-scripting-required.85850/](url) 

I could also upgrade to the ASUS ZenWiFi Pro ET12 and ditch my ZenWifi ET8 / RT-AXE95Q, but why should I have to spend another ~$430 just for VLAN support and no other performance gains?

### **Additional context**
In addition to VLAN, AsusWRT 5.0 (3.0.0.6 Branch) firmware adds several other useful features like:

1. Add Guest Network Pro:
• Kids WiFi: Create a network for kids that blocks access to adult content and has a schedule to control when the network is available.
• VPN WiFi: Create a VPN network that connects to 3rd party VPN services or with ASUS site to site VPN(https://www.asus.com/support/FAQ/1048281/) to encrypts your internet connection and hides your IP address to protect your online activities from being tracked or monitored.
• IoT WiFi : Create an IoT (Internet of Things) network that blocks malicious traffic* and only allows 2.4GHz devices to connect.
• Guest WiFi: Create a guest network with a WiFi schedule and access rights to control when and how guests can use the network.
2. Add VLAN in LAN settings, supports Trunk, Access mode with VLAN-only-network.
3. Add Multi-services WAN, create multiple profiles on a single WAN interface.
4. Add port isolation and AP isolation to restrict device access to each other.
5. Add Auto USB WAN backup
6. Add iPhone tethering

VLAN is by far the MOST IMPORTANT, but these other features would be nice as well. It looks like version 3.0.0.6.102_21514 is the latest "stable" release of AsusWRT 5.0, so maybe that could work too?

gnuton, 2023-11-08:
Merlin is based on 3.0.0.4. Once we will move the 3.0.0.6 we will evaluate which of these feature will be ported.
ATM I do not have access to 3.0.0.6 source code and it would be stupid to implement something which will be replaced in the next major release.

Insan1ty0ne, 2023-11-08:
> Merlin is based on 3.0.0.4. Once we will move the 3.0.0.6 we will evaluate which of these feature will be ported. ATM I do not have access to 3.0.0.6 source code and it would be stupid to implement something which will be replaced in the next major release.

Yes, I completely agree. No sense in spending the time / effort to back-port features to Merlin 3.0.0.4 that Merlin 3.0.0.6 already has. But, when will the source for Merlin 3.0.0.6 become available? Do you have any sort of idea? Asus has had a "Beta" version out for nearly an entire year now, and they have had a stable version out for the RT-AX88U Pro for nearly 6 months. That is quite a long time for a firmware to be in "Beta".

gnuton, 2023-11-09:
no plan yet on when 3.0.0.6 will be available on Merlin. ATM I heard that's is not stable yet. The merge it's not straightforward so it will require some work, which hasn't been estimated yet.
Not coming this year for sure on Merlin.

Insan1ty0ne, 2023-11-09:
> no plan yet on when 3.0.0.6 will be available on Merlin. ATM I heard that's is not stable yet. The merge it's not straightforward so it will require some work, which hasn't been estimated yet. Not coming this year for sure on Merlin.

I hope that the Merlin team can get it stabilized so that the source can be made available. Having the 3.0.0.6 code base on older hardware will save a lot of people a lot of money from having to upgrade just to get simple features like VLAN, port / AP isolation, and the ability to create multiple guest networks. I hope it can be done sooner rather than later.

robbbaxley, 2023-11-14:
I second the ability to add/utilize VLANs once it makes since to add that feature

gnuton, 2024-02-24:
https://www.snbforums.com/threads/how-to-use-vlans-on-your-non-pro-asus-router-with-386-or-388-code-no-scripting-required.85850/
for not pro routers