gnuton/asuswrt-merlin.ng #493: TUF-AX3000

qqw1231, 2023-11-13:
Factory settings
When filling in the telecommunications account password WIFI account password, a second page will pop up to fill in the telecommunications account password WIFI password

gnuton, 2023-11-14:
there is somethng not right with your browser

qqw1231, 2023-11-14:
> 你的浏览器有点不对劲寻求帮助和提示

There's no problem with the browser. I have the same problem I mentioned on both my phone and computer

qqw1231, 2023-11-14:
> 你的浏览器有点不对劲寻求帮助和提示

Edge browser and all browsers on my phone have this issue

gnuton, 2023-11-15:
I wil check this.

qqw1231, 2023-11-16:
> 我会检查一下，你能一步一步地描述一下如何做到这一点吗？您也可以添加屏幕截图吗？

[image]

[image]

qqw1231, 2023-11-16:
Chapter 3: The image automatically pops up

qqw1231, 2023-11-16:
Sometimes when filling out a telecommunications account, the second page will automatically pop up

gnuton, 2023-11-16:
Thanks a lot. I wil check it.

qqw1231, 2023-12-31:
> 非常感谢。我检查一下。

Filling in the router password will still cause this problem.

qqw1231, 2023-12-31:
> 非常感谢。我检查一下。

This bug still pops up when filling in the router password.

gnuton, 2023-12-31:
This bug has not been fixed yet.