gnuton/asuswrt-merlin.ng #494: Possible bug with XT8 v1

robbbaxley, 2023-11-14:
**Router Model Affected**
Asus XT8 v1

**Firmware Version Affected**
RT-AX95Q_3004_388.4_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Only 3 of the 4 CPU cores show utilization. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install firmware version: RT-AX95Q_3004_388.4_0-gnuton1
2. Click on Network Map > Status (Tab under system status)
3. View Processor utilization graph
4. See error: Core 1, 2, and 3 show utilization. Core 4 does not. Utilizing Asus firmware, all 4 cores show utilization. Also, the 3 cores are working much harder than I am used to seeing.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

gnuton, 2023-11-15:
Thanks for reporting this may be a regression. I will look at it.

robbbaxley, 2023-11-16:
I don't know if this makes any difference, but I have 5 routers in AP-only mode, all hardwired to the switch. This is a mix of 3 XT8's and 2 XT9's.

gnuton, 2023-11-16:
I do not recall if this was a visualization issue or just the way the cores are scheduled for this hardware.
I will check and get back.
In any case I won't be worried too much, because if it's a visualization bu it will be fxed. if this is the standard way for this router to work.. then it's by design.

robbbaxley, 2023-11-16:
If you wish, I can put the original Asus firmware back on the hardware and take screenshots or a short video of all 4 cores being utilized. Since I am only using these as AP's I am not noticing any throughput issues and no user has complained. 

I am also able to assist you in tracking down the issue if you want. I am a hardware guy, and don't know software very well but I am happy to help!

gnuton, 2023-11-16:
I think the CPU usage widget is not in the stock firmware. 
If there is a problem is a visualization problem if you ssh into the router and you install enteware you should be able to check the cpu usage with htop command

gnuton, 2023-11-16:
I have not investigated this yet, but I have noticed TUF-AX3000 has same issue.
The last bar doesn't show any activity sign.
BTW http://router.asus.com/Tools_Sysinfo.asp has "CPU Load Average (1, 5, 15 mins)" which shows the average usage and at least on TUF the last core average is not 0

robbbaxley, 2023-11-17:
gnuton, the CPU widget is in the stock Asus firmware. Would you like me to reload it on one of my AP's and screenshot it for you?

gnuton, 2023-11-18:
This doesn't seem to be an issue.
if you like to get confirmation feel free to check the stock firmware too.
I will close the ticket. 
https://www.snbforums.com/threads/ax56u-386-1-cpu-core-4-not-being-used.70017/

I will reopen if the issue if confirmed, but for not it's not

closed 2023-11-18 by gnuton