gnuton/asuswrt-merlin.ng #496: RT-AX82U v2 - Cake QOS is not running

codokie, 2023-11-16:
**Router Model Affected**
Model: RT-AX82U v2

**Firmware Version Affected**
3004.388.4_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
I don't think so.

**Describe the bug**
Cake QOS does not seem to do anything when enabled.
I get 'C' in bufferbloat tests such as the ones at DSLReport, and VoIP is very bad when another device starts downloads - the same behavior is exhibited when QOS is disabled.
Even the bandwidth values (80 down/3.6 up in my case) specified at the QOS page were not respected, and speed tests would show values above that limit.
I have examined the system log and did not find any errors mentioning cake-qos.
On a side note, Traditional QOS seems to be working for the most part, but it isn't a very good QOS solution.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Adaptive QOS' -> 'QOS' tab
2. Set upload and download bandwidth to 90% of the speed provided by ISP and enable Cake QOS (router is rebooted)
3. Go to 'CakeQOS-Merlin' tab (installed via the script [here](https://github.com/ttgapers/cakeqos-merlin) - note that the problem persists even if the addon is not installed)
4. See error under status: "Cake is not running". Speed tests show values higher than the ones set in step 2

**Expected behavior**
Cake should be reducing bufferbloat and not use more bandwidth than specified.

gnuton, 2023-11-16:
thanks for reporting I will check this out

swwwong, 2023-12-31:
Router Model Affected
Model: DSL-AX82U

Firmware Version Affected
3004.388.4_0-gnuton1

Is this bug present in upstream Merlin releases too?
no idea

closed 2025-08-20 by codokie