gnuton/asuswrt-merlin.ng #500: ZenWiFi XT8 v1 only 3 of 4 CPU cores in use [Updated]

robbbaxley, 2023-11-26:
**Router Model Affected**
Models: [ZenWiFi XT8 v1

**Firmware Version Affected**
RT-AX95Q_3004_388.4_0-gnuton1_puresqubi

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
A few days ago, I reported that only 3 of 4 CPU cores are being utilized in this firmware version. All cores are being utilized in all releases of the official Asus firmware.  I was doing some additional analysis and testing and found that when I transferred date to a USB drive (SSD) connected to the USB port on the router, 3 of the 4 cores would operate at or near 100% with the 4 core finally showing some utilization but only 3% to 5%. Transfer speed was slow at an average of 33Mb/s.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Network Map'
2. Click on 'Status' tab
3. Watch CPU utilization graph
4. See error

**Expected behavior**
A 4 CPU cores to be consistently utilized and better transfer speed to USB drive.

**Screenshots**
Attached
[image]

dacootmeister, 2023-11-29:
It's long known issue, no fix until now
The 4th core is quieter then the first , but low level it still does something from linux htop:
[image]

robbbaxley, 2023-11-29:
It's odd, though, that all 4 cores are move evenly utilized on the standard Asus WRT firmware and I never saw utilization spike to (and stay at or near 100% utilization. To me, it appears that threading/scheduling is somehow different.

jazha, 2023-11-30:
While transferring files internally on my network, from a wired source to a wifi destination, I get life from core 4 in gui.

[image]

robbbaxley, 2023-11-30:
This is so odd! 

Can I ask which version of firmware you are using? Do you know which version of hardware you have? Mine is v1 and I see utilization similar to what you are showing, but only when using the standard firmware. I have verified standard core utilization on Asus 3.0.0.4.388.23285 firmware.

jazha, 2023-11-30:
Im on a V1 aswell.

I run 3004.388.4_0-gnuton1

Not sure how to extract the exact numbers, but its the only one gnuton released for 3004 series of FW I believe

robbbaxley, 2023-11-30:
[image]

I took this capture just now. The only time I see any utilization at all on Core-4 is when I am transferring data to the USB disk over the network. The first 3 cores work at nearly 100% during the transfer but Core-4 is barely utilized. I have 5 devices, all AP's and all connected over Wired backhaul (1 device is currently offline due to a backhaul failure). I have 3 XT8's and 2 XT9's. All version 1's. The XT9's are using the default Asus firmware.

jazha, 2023-11-30:
Could be core is locked to specific tasks, like running ethernet/wifi connections etc. I dont know enough to figure that out. I wouldnt worry about low load unless you have issues with the router not working properly. Someone more familiar with the workings could probably shed some light on the 4th cores tasks?

robbbaxley, 2023-11-30:
Here is another test I completed just now. First screen capture is after plugging in a USB drive. Nothing else. Notice how CPU utilization increased though I don't know why. In the second screen capture, it shows data transfer over the wired network to the USB drive. CPU utilization nearly maxes out the first 3 cores. The 4th core finally shows some utilization but only just.

[image]

[image]

robbbaxley, 2023-11-30:
I should add, my only concern is that the data transfer rate over the wired network to the USB drive is much slower than expected.

jazha, 2023-11-30:
What type of drive do you have connected? Maker/model/connection/portable/powered/size/file system etc? 

Could be something as simple as different drives just work differently unfortunately.

robbbaxley, 2023-11-30:
In the very first screen capture, for that test I used a 7.68TB Micron 5300 Pro SSD (550MB/s write speed). In the most recent test, I used a PNY 512GB Elite-X Fit USB 3.1 Flash Drive (200MB/s write speed).

Both drives were had a new partition created and formatted NTFS in Windows 11 Pro. I did attempt to use the PNY drive as exFAT but the Asus router wouldn't use it with that file system.

robbbaxley, 2023-11-30:
Yeah, I am stumped too. I tested my network card speed and it maxes out at just over 2.1Gb/s so I am confident that's not the bottleneck. I am pretty sure that the issue lies somewhere in the firmware. Since the 3 working cores max out I am reasonably confident that there is an issue with the USB and/or CPU threadding.

soelu, 2023-12-08:
[image]
I am running the stock firmware still and see the same effect. So i assume this is not a Merlin but a Asus Stock Firmware "Issue"

Dodgydrains, 2023-12-09:
[image]

Seems fine to me on stock 23285 FW

robbbaxley, 2023-12-09:
@soelu, are you on hardware v1 or v2? I have v1 and, like @Dodgydrains, the cores operate properly on stock firmware for me.

Dodgydrains, 2023-12-09:
@robbbaxley I am also on V1 hardware and stock FW and cores seem fine on it.

Dodgydrains, 2023-12-10:
[image]

This seems to be working for me on latest 3004.388.5_0-gnuton0_beta1, I did dirty flash from stock 23285 FW.

gnuton, 2023-12-12:
The load is not spread across all cpus, since cpu affinity kicks in to keep some processes in some threads. This has been manually fine tuned by ASUS and it's what you see in all ASUS routers.

This has been already discussed a while ago and RMerl has replied to this too with some tests.
https://www.snbforums.com/threads/rt-ac66u_b1-not-using-2nd-cpu-core.39670/

soelu, 2023-12-12:
So I guess this is a non-issue for the Merlin Firmware then?

gnuton, 2023-12-12:
@soelu according to what I have read in the post there were not advantages. if you have specific applications which you wnna fine tune for using more threads you may try taskset
```bash
taskset -p 1 "$pid";
```
You can get the pid of your app running "ps" from the command line
and you can replace 1 with the bitmask associated to the cores you want to use.

robbbaxley, 2023-12-13:
I can also confirm that, as @Dodgydrains noted, the ZenWifi XT8 Beta firmware version RT-AX95Q_3004_388.5_0-gnuton0_beta1_puresqubi, is working much better as it pertains to thruput to the USB attached storage. 

I would like to note the the UL and DL appear to be backwards to me? Any thoughts?

Dodgydrains, 2023-12-13:
> I can also confirm that, as @Dodgydrains noted, the ZenWifi XT8 Beta firmware version RT-AX95Q_3004_388.5_0-gnuton0_beta1_puresqubi, is working much better as it pertains to thruput to the USB attached storage.
> 
> I would like to note the UL and DL appear to be backwards to me (I am uploading files to the USB drive from a PC). Any thoughts?
> 
> ![Web capture_13-12-2023_111135_10 0 1 200](https://private-user-images.githubusercontent.com/150712747/290263257-6fd531f8-975d-4fd6-9157-762e3f22a3e5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1MDMyNTksIm5iZiI6MTcwMjUwMjk1OSwicGF0aCI6Ii8xNTA3MTI3NDcvMjkwMjYzMjU3LTZmZDUzMWY4LTk3NWQtNGZkNi05MTU3LTc2MmUzZjIyYTNlNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxM1QyMTI5MTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMjA4MzYwZDI4MDE2MjJkMDFhM2I5ZGNiYWMwM2M2ZmQzZDhmNmU4ZTQ4NjdkMmQxZWZlMGI2OTQyNWUwZWVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bVdgh2fZAJz1q0bLF7EbXGzvgmL9qhadA0-8W11i7sc)

Mine seems fine, but I did dirty flash over official FW 23285

robbbaxley, 2023-12-14:
@Dodgydrains, I just checked and also verify that the Ul/DL looks correct when in internet download/streaming/speed-test. So the graph is probably correct and it just looks backwards to me.

Now if we could just figure out WHY on the Merlin firmware 3 cores are utilized heavily when, in the Asus released firmware, the load is much more evenly balanced across all 4 cores. At least USB transfers are working quite a bit better with this Beta firmware.

robbbaxley, 2024-01-04:
I wanted to post an update regarding the new firmware version RT-AX95Q_3004_388.5_0-gnuton1_puresqubi.w. 

Still showing 3 of 4 cores with high CPU utilization during file transfers to a USB drive connected to the primary node. The 4th core shows o% to 4% utilization during data transfer to USB storage. This is the same as with the previous firmware version.

I set the USB version to both USB 2.0 and USB 3.0 in the primary node (where the drive was connected) and tested with no change in transfer speed. All nodes connected via hardwire at 2.5Gb/s. Computers used to test was 1 Windows 11 desktop connected via 1Gb/s hardwire and 1 Linux laptop connected via wireless. There was no difference in CPU utilization on the primary node with the connected storage drive and no difference in data transfer rates.

LarryRosen108, 2024-01-04:
I have seen similar results with higher node 4 % but without any real
impacts to usage of my 2 node XT8 environment. My transfer was done using
winscp from a laptop with 1gb Ethernet connection directly to my main node
that contains the USB 3.0 stick. This is on the latest .5 release.

_______________________________________________________
Sent from my mobile device so please pardon autocorrect and brevity

swwwong, 2024-01-12:
@robbbaxley 
My DSL AX 82U also have the same problem .By default many processes  only select to run at core 1 and 2 .

I use scMerlin in ssh and look at the task.
 (amtm-->j3-->c)
add "PROCESSOR "in to Activate Columns. (F2-->Screens )to see task are working on cpu [?] .

slot by CPU% ,select tasks that in high usage ,type A ,left side will show  task  are assigned [X] CPU (1,2,...).Not assigned [ ].
 Press space to select or un-select. and enter to save .

×some  task couldn't change working cpu.

×make sure all task have assigned at least one cpu ,otherwise system may crash .

swwwong, 2024-01-12:
@robbbaxley 

Here show why cpu 2 or in your case cpu 3 at idle .
Most task only set to use core 0 and 1 ; 0,1 and 2 in yours.
you can manaul adjust some task to only core 3 .

[image]

[image]

doomwithdon, 2024-01-15:
so this will always be an issue on gnuton builds only?

swwwong, 2024-01-15:
@doomwithdon 

I think this issue is from the asus-wrt  fiwmware.

Befor install gnuton builds,CPU 3 stays idle.

it only use cpu 3 will used  when I config an  second open vpn client in gnuton builds.

[image]

davidpalmeruk, 2024-01-27:
I came to add a vote to support XT9 and spotted this - my XT9 has this underutilised CPU4  so can only agree with comments that it seems to be just part of the Asus stock firmware.