gnuton/asuswrt-merlin.ng #502: UL/DL values/labels are reversed

ShakeforProtein, 2023-11-27:
**Router Model Affected**
Models: DSL-AX82U (though I assume others also)

**Firmware Version Affected**
3004.388.4_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
No idea

**Describe the bug**
Upload/Download speeds are reversed in System Status

**To Reproduce**
1. Goto > General > Network Map > System Status
2. Observe listed UL / DL speeds

**Expected behavior**
UL speed and DL speed values should be the opposite.

**Screenshots**
[image]

ShakeforProtein, 2023-11-27:
Must be, my apolgies.

gnuton, 2023-12-03:
can I close this?

ShakeforProtein, 2023-12-03:
Sure if you don't care about the issue. It doesn't really bother me, just wanted to make sure it was known.

gnuton, 2023-12-10:
@ShakeforProtein My apologies I misread the comments and I thought this was not a bug.
I do not have a DSL-AX5400 variant so i am not able to test that device, but I do see your system status is also missing the AURA and Wireless tab, isn't it? there is something wrong with your UI.
Am i right?

ShakeforProtein, 2023-12-10:
Both Aura RGB and Wireless tabs exist, however require hovering over where they should be before they become visible. Also the Aura RGB tab has a white background with white text.

Using Opera (The browser I typically use) I get the following (see pictures). Having tested in Edge and Firefox the css still can't be found but the correct styles are used, the tabs are visible and the aura rgb tab looks normal.

[image]

[image]

[image]