gnuton/asuswrt-merlin.ng #503: 🌟 Support for ZenWiFi ET8 - RT-AX5400 🌟

gnuton, 2023-12-01:
*This thread is about supporting RT-AX5400.
If you see this issue here is because I have decided to support this model.

So, feel free to ask questions here about this topic.
I will keep the status updated.

Status:
- [x] received GPL
- [x] received development unit
- [] Merged GPL to feature-branch
- [] Initial testing & Bugfixing phase
- [] Public pre-alphas
- [] Alpha released
- [] collecting testing reports  & Bugfixing phase
- [] Beta released
- [] collecting testing reports  & Bugfixing phase
- [] Stable released

gameinn1, 2023-12-26:
Is there any sort of forum or whatever where people can vote on what router should get support?

gnuton, 2023-12-28:
the prebuild coming within hte GPL do not have compatible prebuilds with the upstream ones provided by Eric, since my models are based on [BCM6756](https://www.broadcom.com/products/wireless/wireless-lan-infrastructure/bcm6756)(ARMv7) and all Eric's SDK 5.04 models are based on [BCM4912](https://www.broadcom.com/products/wireless/wireless-lan-infrastructure/bcm4912) (ARMv8)
Issues:

- src-rt-5.04axhnd.675x/bcmdrivers/opensource/net/enet/impl7/bcm_enet.o
- src-rt-5.04axhnd.675x/bcmdrivers/opensource/phy/prebuilt/phy_drv.o

gnuton, 2024-01-01:
any tester?

Imastupidsalmon, 2024-01-03:
> any tester?

I would enjoy the opportunity to test for this device!  I have (2) RT-AX5400 REV B1 new, in-box and (1) RT-AX3000 with Merlin on it in use as my primary home router at the moment.

CMJ

gnuton, 2024-01-03:
@Imastupidsalmon you can find the pre alpha image here. This image has not been tested and **IT CAN SOFT BRICK YOUR ROUTER**.
So please before flashing download the [asus recovery tool](https://www.asus.com/support/FAQ/1000814/)
https://app.circleci.com/pipelines/github/gnuton/asuswrt-merlin.ng/1726/workflows/5671c5d6-91a1-4efc-8e8e-34a28d2f7188/jobs/6105/artifacts

If you do not feel enough brave to test this pre-alpha image it's fine.. I will test it next week once I get the device in my hands.

Imastupidsalmon, 2024-01-04:
I don't mind risking a soft brick at all. Now that I grabbed the restoration utility and stock firmware, and now that I have downloaded your pre-alpha, I will trying flashing it and share the results shortly. Thank you for your work on this and letting me give it a whirl!
CMJ

Imastupidsalmon, 2024-01-04:
No problems so far! Everything I've checked operates as it should, and well, as far as I can tell so far. I just swapped this ax5400 with your pre-alpha on it into place as my home's primary wireless router so that I can give it a more realistic load/test. But so far it's running great, will keep you posted if that changes. Thank you, sir! And great work! If you need me to try something on my end, do let me know and if I can I will. 

[image]

gnuton, 2024-01-07:
Here another tested by myself pre-alpha build
https://output.circle-artifacts.com/output/job/20e496ac-68f8-4d76-bf83-cf6b99a8814b/artifacts/0/workspace/release/RT-AX5400_3004_388.6_alpha1_nand_squashfs.pkgtb

radugrecu97, 2024-01-19:
Question, does support for this model mean that XD6 will be technically supported? The [product page](https://www.asus.com/networking-iot-servers/whole-home-mesh-wifi-system/zenwifi-wifi-systems/asus-zenwifi-xd6/) says they're a pair of AX5400 routers.

I don't know if it's just a different case.

Imastupidsalmon, 2024-01-22:
[image]

[image]

12 days without a reboot. 17 devices connected. Speed is fantastic. Thank you @gnuton ! It's working great for me!

CMJ

closed 2024-03-08 by gnuton

gnuton, 2024-03-27:
@Heiliger-Gral it s stable enough

Pink210, 2024-05-05:
sorry for bother :(
this topic is about this RT-AX5400:
https://www.asus.com/networking-iot-servers/wifi-routers/asus-wifi-routers/rt-ax5400
or this one(DSL-AX5400):
https://www.asus.com/networking-iot-servers/dsl-ax82u/
?

gnuton, 2024-05-07:
Rt-ax5400

Hollywood25201, 2024-05-19:
@gnuton
New to firmware installs on routers, was wanting to see where I can download the firmware for this. I can't find it on the main site (https://www.asuswrt-merlin.net/). I found it in the release page of github but only found "[RT-AX5400_3004_388.6_2-gnuton1_nand_squashfs.pkgtb](https://github.com/gnuton/asuswrt-merlin.ng/releases/download/3004.388.6_2-gnuton1/RT-AX5400_3004_388.6_2-gnuton1_nand_squashfs.pkgtb)"
On all the tutorials online it says it's a .w file, but I don't want to brick my router.
Could I get some guidance?

gnuton, 2024-05-20:
@Hollywood25201 the flashing tool checks for the firmware signature. You should allow you to flash garbage.
Even in the worsrt case you can recover but yes the last link is the rcorrect one

LanGsA0, 2024-10-18:
Hi~!! I know there are V1 and V2. Which version is the AX5400 here?

SMOL25, 2025-03-31:
https://amzn.eu/d/1B1xN4H

Is this the router of this topic?

zarif98, 2025-05-04:
Has this been merged? I see it's closed but I would like to have merlin installed on the 5400.

Ryder17z, 2025-05-31:
> Has this been merged? I see it's closed but I would like to have merlin installed on the 5400.

same. but also, what hardware version?
Mine is marked "A" instead of "v1"  - are the two synonyms in this case?

Bugrakca, 2025-09-21:
I want to buy RT-BE58U. Is it going to supported I don't want to waste money. These are not cheap. Or should I buy AX5400?