gnuton/asuswrt-merlin.ng #505: Strange CPU behavior

Barmagler, 2023-12-09:
**Router Model Affected**
Models: TUF-AX5400

**Firmware Version Affected**
3004.388.5_0-gnuton0_beta1, 3004.388.4_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
There is very strange situation.
I had stable version 388.2_2_0-gnuton1 but today suddenly updated to 3004.388.5_0-gnuton0_beta1.
After reboot noticed that CPU1 and CPU2 make spikes to 25-35% with no traffic at all.
Ok, I made update to stable version, 3004.388.4_0-gnuton1 but situation still same.

I belive than I was at 388.2_2_0-gnuton1 it was 5% CPU maximum.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Network map'
2. Click on 'System status -> Status'
3. Watch CPU graphic

**Expected behavior**
I really don't understand: there is problem with my router or 388.2_2_0-gnuton1 has problem with CPU graphic?

**Screenshots**
[image]

gnuton, 2023-12-10:
can you install htop and report back the processes causing these spikes?

Barmagler, 2023-12-10:
Gotcha!
/usr/sbin/awsiot

[image]

Barmagler, 2023-12-13:
Actually it's much more better now.
My bad, I really forgot make reset.

Now only two processes make 5-7% - networkmap and cfg_server

[image]

closed 2023-12-13 by Barmagler