gnuton/asuswrt-merlin.ng #506: LED on/off toggle not working for RT-AX92U

gnuton, 2023-12-10:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
3004.388.4-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Led on/off toggle not working

gnuton, 2023-12-10:
fixed with 7a559159eb

gnuton, 2023-12-10:
test images here https://app.circleci.com/pipelines/github/gnuton/asuswrt-merlin.ng/1704/workflows/b4730fc7-5e01-47bc-bf01-cbc51ebca5c5/jobs/5855

gnuton, 2023-12-10:
test images here https://app.circleci.com/pipelines/github/gnuton/asuswrt-merlin.ng/1704/workflows/b4730fc7-5e01-47bc-bf01-cbc51ebca5c5/jobs/5855/artifacts

ExplorerNL, 2024-07-10:
Same issue here, latest stable version for AX92U, LEDs on node does not remember the settings when turned off and on. LEDS turn ON after boot even when settings shows OFF

ExtremeFiretop, 2024-07-10:
@gnuton 

@ExplorerNL  noticed this issue in discord and I validated it, we are both ready to test for you are your earliest convenience @gnuton 

I noticed you had test builds up, but I can't seem to access them anymore (almost a year later)