gnuton/asuswrt-merlin.ng #507: TUF-AX3000

qqw1231, 2023-12-13:
[image]

Up and down not centered
[image]

572289049, 2023-12-15:
版本号是？如果你不喜欢，可以使用TUF的皮肤版本。

qqw1231, 2023-12-15:
> 版本号是？如果你不喜欢，可以使用TUF的皮肤版本。

3004.388.5_0-gnuton0_beta1

gnuton, 2023-12-15:
@qqw1231 do you get the same issue changing the language to english?

micron888, 2023-12-16:
TUF-AX5400 ok with both english and chinese
[image]
[image]

qqw1231, 2023-12-16:
> TUF-AX5400中英文都可以 ![image](https://private-user-images.githubusercontent.com/59114523/290975293-1870bf0d-9e7b-4daf-a629-caf9b3eeadaa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI3MTc2MTUsIm5iZiI6MTcwMjcxNzMxNSwicGF0aCI6Ii81OTExNDUyMy8yOTA5NzUyOTMtMTg3MGJmMGQtOWU3Yi00ZGFmLWE2MjktY2FmOWIzZWVhZGFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE2VDA5MDE1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA5NGM5MTUwYzY0ZTc5Yjg1ZTViN2U4MzQxZjVjYTg3ZDNkNTk0NTZmOTM1YTY2NDFkMmIwYTZkMmQxMTg0MWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qLCvhpV5-V3HwUCOiJDwJCcr-S_8xrNoxVwqZSpOEys) ![image](https://private-user-images.githubusercontent.com/59114523/290975300-eee24e37-dab5-45e4-94d9-65998db265e7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI3MTc2MTUsIm5iZiI6MTcwMjcxNzMxNSwicGF0aCI6Ii81OTExNDUyMy8yOTA5NzUzMDAtZWVlMjRlMzctZGFiNS00NWU0LTk0ZDktNjU5OThkYjI2NWU3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE2VDA5MDE1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVkM2E5NzAxZGM4Yzc1MmQwYzliMTEyZWFhNjg4ZjZmOGI4OGZkMjVlMDI4ZWJhOGMzYjYyYzE3NzA4ODFhM2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.t-6EUl3qeQLq47-Ev8xtAu_Aj1AnCu5VQnms8g0jNvY)

Please provide a complete picture and let me see where there are spaces in Tufax 3000.

qqw1231, 2023-12-16:
> TUF-AX5400 ok with both english and chinese [...]

Where is a large space in the first picture, I marked it with a circle.

qqw1231, 2023-12-16:
> @qqw1231你会遇到同样的问题吗？

The circle I drew means there is a lot of emptiness

micron888, 2023-12-17:
35devices connected :D no emptiness
[image]

gnuton, 2023-12-18:
I cannot reproduce it. Can you test this in an incognito browser?

tony8077616, 2023-12-24:
@gnuton I also have the same problem with the layout doesn't with enough space to show
Use with 3004.388.5_0-gnuton0_beta1

Microsoft Edge 120.0.2210.91
[image]
[image]

Same with Google Chrome canary 122.0.6203.0
[image]
[image]

swwwong, 2024-01-05:
@tony8077616 

This UI  ''BUG'' also same in the asuswrt factory software.

tony8077616, 2024-01-05:
> @tony8077616 
> 
> This UI  ''BUG'' is same in the asuswrt factory software,it have a fix UI size

Got it, Thanks.

gnuton, 2024-02-23:
fix 2588d458b97d21e4900f7f5f8d50a3127916f9ee

closed 2024-02-23 by gnuton