gnuton/asuswrt-merlin.ng #509: AX82U v2 Control Channel Issue

Qasper93, 2023-12-19:
**Router Model Affected**
Models: (eg: DSL-AX82U/RT-AX82U/...)
RT-AX82U V2
**Firmware Version Affected**
3004.388.4_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes/No
N/A
**Describe the bug**
Unable to change control channels on either 2.4ghz/5ghz .
**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Wireless" under Advanced Settings
2. Under "Control Channel" change from "Auto" to designated channel (e.g. 60)
3. Click "Apply"
**Expected behavior**
To update Control Channel to designated channel - this does not and reverts back to "Auto" every time.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section
[image]

gnuton, 2023-12-19:
Can you please reset to factory and try again?

Qasper93, 2023-12-19:
> Can you please reset to factory and try again?

Hi, I have tried factory resetting and this did not make a difference.

huwu2705, 2023-12-31:
I just updated to version 3004.388.5, the problem still exists

codokie, 2024-01-01:
Have you tried turning off Smart Control? I am able to change control channel and channel bandwidth if Smart Connect is turned off.
It may also be possible to do so without disabling Smart Connect if you access the relevant setting using the mobile app instead of web GUI

saber-nyan, 2024-05-15:
Also happens on RT-AX58U without Smart Connect.