gnuton/asuswrt-merlin.ng #511: ZenWiFi XT8 / RT-AX95Q HW v2 Support

Verrou, 2023-12-23:
**Is your feature request related to a problem? Please describe.**
It appears the support for ZenWiFi XT8 / RT-AX95Q is v1 - when I try using the FW it simply freezes the firmware/router page.

**Describe the solution you'd like**
Clarify if v2 is supported or add support if able

**Describe alternatives you've considered**
N/A

**Additional context**
Official version from ASUS here
https://dlcdnets.asus.com/pub/ASUS/wireless/ZenWiFi_XT8/FW_ZENWIFI_XT8_V2_300438823285.zip?model=ASUS%20ZenWiFi%20AX%20(XT8)

gnuton, 2023-12-23:
You cannot run V1 firmware on V2 Hardware, Simply because the hardware is different and so the software. The V2 Firmware is very very different from V1. It s built from a different broadcom SDK.
V2 is not supported.

Verrou, 2023-12-23:
Thanks @gnuton yep I understand you can't run v1 fw on v2 hw, was just seeking clarification that the fw available was v1. Can support for v2 be added in the future? Cheers :)

gnuton, 2023-12-23:
Maybe. If you get into discord and you check in the news channel you can see the upcoming devices. V2 won't come in H1 for sure.

closed 2023-12-23 by gnuton

gnuton, 2023-12-27:
@Verrou no plans to add V2 support as for now. So not coming at least in the first half of 2024