gnuton/asuswrt-merlin.ng #512: DSL AX-82U CakeQOS not workinng

swwwong, 2024-01-01:
Router Model Affected
Model: DSL-AX82U

Firmware Version Affected
3004.388.4_0-gnuton1 , 3004.388.5_0-gnuton1

Is this bug present in upstream Merlin releases too?
I don't think so.

Describe the bug
Cake QOS works fine for first 30 sec to about a minuite in bufferbloat test.
Than it will just stop working sand shows ''Cake is not running'' in the CakeQOS-Merlin tabs.
and it get F in the result.
LATENCY:
Unloaded: 16 ms
Download Active:+9 ms
Upload Active:+430 ms

To Reproduce
Go to 'Adaptive QOS' -> 'QOS' tab
Set upload and download bandwidth to any value (1-100%)of the speed provided by ISP and enable Cake QOS 
(router is rebooted)
uninstall and reinstall CakeQOS-Merlin
try without install CakeQOS-Merlin.
(installed via the script [here](https://github.com/ttgapers/cakeqos-merlin) ,run cake-qos status general,shows ''Cake is not running'' .

Expected behavior
Cake should be reducing bufferbloat and not use more bandwidth than specified.

gnuton, 2024-01-01:
any log?

Yakumounr, 2024-01-01:
I run a DSL-AX82U, just chipping in to say this was never an issue here, perhaps a dirty firmware install?

(Also cake-qos has a v2.1.2 update which isn't on the releases page but it will update via the button at the top, but that shouldn't be related to this issue).

gnuton, 2024-01-01:
@Yakumounr the fact that  it works only for 30 seconds means there is something else breaking this functionality. The log may help here to pin point the issue with some settings or installed scripts.
Suggestion: get a clean environment. Format the router and do not apply previous setting backup. Just add the bare minimum settings little by little and see if it works and what breaks it

swwwong, 2024-01-01:
@gnuton 

The ''CakeQOS-Merlin: Not running...'' dosn't show on the System Log . It only shows up on the CakeQOS-Merlin Tab or enter the commemd ''cake-qos status general'' in the SSH.

here some log shows when Cake Qos not working:

Jan  1 22:25:17 kernel: [77415.925058] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=205.210.31.33 DST=86.163.83.27 LEN=44 TOS=0x00 PREC=0x00 TTL=250 ID=54321 PROTO=TCP SPT=54518 DPT=5050 SEQ=1650362543 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (020405B4) MARK=0x8000000 
Jan  1 22:25:22 kernel: [77421.280254] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=45.134.26.45 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=244 ID=53911 PROTO=TCP SPT=57676 DPT=58887 SEQ=1716676491 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:25:29 kernel: [77427.944224] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=208.100.26.241 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=243 ID=41958 PROTO=TCP SPT=46840 DPT=80 SEQ=2741039277 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:25:44 kernel: [77442.462368] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=45.134.26.45 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=242 ID=51887 PROTO=TCP SPT=57676 DPT=59814 SEQ=1725657106 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:25:51 kernel: [77448.918380] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=194.26.135.247 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=245 ID=3932 PROTO=TCP SPT=58818 DPT=33945 SEQ=1316739294 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:26:00 kernel: [77458.281473] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=109.205.213.90 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=244 ID=12494 PROTO=TCP SPT=47127 DPT=21200 SEQ=1543581957 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:26:07 kernel: [77465.610242] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=176.241.83.187 DST=86.163.83.27 LEN=119 TOS=0x00 PREC=0x00 TTL=53 ID=61589 DF PROTO=UDP SPT=33372 DPT=8000 LEN=99 MARK=0x8000000 
Jan  1 22:26:31 kernel: [77488.843864] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=220.147.125.223 DST=86.163.83.27 LEN=44 TOS=0x00 PREC=0x00 TTL=47 ID=55446 PROTO=TCP SPT=26703 DPT=23 SEQ=1453544219 ACK=0 WINDOW=35623 RES=0x00 SYN URGP=0 OPT (02040218) MARK=0x8000000 
Jan  1 22:26:34 kernel: [77492.695321] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=212.70.149.86 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=242 ID=54321 PROTO=TCP SPT=37808 DPT=35748 SEQ=178156793 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:26:41 kernel: [77499.712837] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=162.243.143.21 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=244 ID=54321 PROTO=TCP SPT=56929 DPT=55554 SEQ=1824193868 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:26:48 kernel: [77506.277971] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=152.89.198.71 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=243 ID=5832 PROTO=TCP SPT=50718 DPT=1083 SEQ=2714780136 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:27:00 kernel: [77517.814899] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=77.90.185.191 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=245 ID=13105 PROTO=TCP SPT=53863 DPT=909 SEQ=1164534695 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:27:02 kernel: [77520.055607] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=109.205.213.90 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=244 ID=51738 PROTO=TCP SPT=47127 DPT=23583 SEQ=2350302494 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:27:13 kernel: [77531.678300] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=77.90.185.146 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=245 ID=37646 PROTO=TCP SPT=55588 DPT=20004 SEQ=3719597617 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:27:39 kernel: [77557.410390] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=80.94.95.226 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=247 ID=37732 PROTO=TCP SPT=48025 DPT=8443 SEQ=3153256580 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:27:54 kernel: [77572.924611] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=152.89.198.75 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=244 ID=40763 PROTO=TCP SPT=50740 DPT=1108 SEQ=328910964 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:28:02 kernel: [77581.136113] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=80.94.95.226 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=247 ID=42819 PROTO=TCP SPT=48069 DPT=8080 SEQ=3317951432 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:28:10 kernel: [77588.832176] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=77.90.185.186 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=245 ID=35937 PROTO=TCP SPT=53830 DPT=33789 SEQ=1077894279 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:28:13 kernel: [77591.391378] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=109.205.213.90 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=244 ID=23910 PROTO=TCP SPT=47127 DPT=32844 SEQ=952405553 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:28:21 kernel: [77599.390568] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=77.90.185.185 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=245 ID=45371 PROTO=TCP SPT=53819 DPT=33892 SEQ=3525731908 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:29:05 kernel: [77643.748354] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=91.92.245.222 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=247 ID=54321 PROTO=TCP SPT=55532 DPT=80 SEQ=813150229 ACK=0 WINDOW=6 [... cut ...]
Jan  1 22:35:46 kernel: [78045.557928] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=109.205.213.90 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=243 ID=28941 PROTO=TCP SPT=47127 DPT=25163 SEQ=4182982749 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:35:57 kernel: [78056.220833] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=77.90.185.128 DST=86.163.83.27 LEN=40 TOS=0x00 PREC=0x00 TTL=245 ID=54321 PROTO=TCP SPT=35368 DPT=63333 SEQ=239038290 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 MARK=0x8000000 
Jan  1 22:35:58 kernel: [78057.045798] [BLOCKED - INBOUND] IN=ppp0 OUT= MAC= SRC=167.94.138.135 DST=86.163.83.27 LEN=74 TOS=0

swwwong, 2024-01-02:
I think I will just leave it for now,because it just the CakeQOS not working ,I can still use the FlexQoS to fix the bufferbloat problem. 

I will just leave it unit my home broadband been upgrade and see.

thanks for support.

winter2222, 2024-01-02:
The legacy CakeQOS-Merlin v1.0 (which is build in fw) is working fine for me, I think what is causing the issue is when its uninstalled and installed CakeQOS-Merlin v2 via script above.

swwwong, 2024-01-12:
Maybe I have Accidentally fix the Cake QOS not working problom.

use scMerlin -->View running processes-->change command:bcmsw_rx and bcmxtm_rx from cpu 0,1 to 2 only.

closed 2024-01-13 by swwwong