gnuton/asuswrt-merlin.ng #513: AMTM in a loop, cant access menu items

dacootmeister, 2024-01-01:
**Router Model Affected**
Models: 	ZenWiFi_XT8 (base model: RT-AX95Q)

**Firmware Version Affected**
RT-AX95Q_3004_388.5_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
ATMT menu in a loop, can't access items to config my mesh router

**To Reproduce**
Steps to reproduce the behavior:
1. Flashed over my previous Gnton build which was RT-AX95Q_3004_388.5_0-gnuton0_beta1
2. Updated ATMT from 4.0 to 4.1, then the issue occurs

**Expected behavior**
AMTM behaving normally
Downgrade AMTM might do the trick as well. 
Reflashing firmware did not

**Screenshots**
cant upload video with .mov extension unfortunately

gnuton, 2024-01-01:
is this an AMTM 4.1 bug?

dacootmeister, 2024-01-01:
> AMTM 4.1 bug

hi Antonio, I've searched for it, no-one reported it until now... so can't tell whether it's build related or tool related.
Only way is to revert back to stock FW, restore from backup and check which of the 2 it is.

dacootmeister, 2024-01-01:
[IMG_0106.zip](https://github.com/gnuton/asuswrt-merlin.ng/files/13806246/IMG_0106.zip)
this zipfile worked to upload & demonstrates what I encounter

gnuton, 2024-01-02:
I do not get why you switch to stock firmware to "fix" this.
It's like installing a different operatating system version if an application on your PC doesn't work.

amtm is a bash script and since you can update it, you can also revert back or check what's the difference between the 4.0 and 4.1 and see what's the problem.
There is very high chance that FACTORY RESETTING or just /jffs formatting will fix this

dacootmeister, 2024-01-02:
> e between the 4.0 and 4.1 and see what's the problem.

> I do not get why you switch to stock firmware to "fix" this. It's like installing a different operatating system version if an application on your PC doesn't work.
>
> amtm is a bash script and since [...]

Well you have to try something and start somewhere. From scratch is the best approach imho, but I can also follow the factory setting suggestion you are providing.
Thanks!
Will try today and revert back to this thread with results.

dacootmeister, 2024-01-02:
Update: restoring a backup of the  JFFS partition I did half a year ago fixed the AMTM issue for now.
Case closed :-)

dacootmeister, 2024-01-02:
fdy

closed 2024-01-02 by dacootmeister