gnuton/asuswrt-merlin.ng #517: Connected Client List Not Updating on Asus ZenWiFi XT8 / RT-AX95Q v1

robbbaxley, 2024-01-05:
**Router Model Affected**
Models: Asus ZenWiFi XT8 / RT-AX95Q v1

**Firmware Version Affected**
Stable: 3004.388.5_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Unknown but not present in Stable: 3004.388.4_0-gnuton1

**Describe the bug**
Client List does not update after the first time.

**To Reproduce**
Steps to reproduce the behavior:
1. Under "General"
2. Click on "Network Map"
3. Scroll down to "View List"
4. Click on "View List"

**Expected behavior**
Client List should update

**Screenshots**
Screenshot attached
[image]

gnuton, 2024-01-06:
Unlikely a bug: most likely  a cache issue of your browser or worst case messed router config.

LarryRosen108, 2024-01-06:
I believe I am seeing the same issue on my 2 node xt8 setup. I can turn off
wifi on my phone and it still shows up. My device count has not changed
even if people come and go.

Larry
------------------------------------------
Sent from my mobile device. Please excuse spelling and brevity.

LarryRosen108, 2024-01-06:
Tonight it seems to be working.

shiang-luong, 2024-01-18:
I don't have that problem with my 2 node xt8