gnuton/asuswrt-merlin.ng #521: Can't upload firmware

manishie, 2024-01-10:
**Router Model Affected**
Models: RT-AX95Q (XT8)

**Firmware Version Affected**
RT-AX95Q_3004_388.5_0-gnuton1_puresqubi.w

**Describe the bug**
I'm unable to upload the new firmware.  I'm currently running stock 3.0.0.4.388_23285, on MacOS 14.1.2 (23B92).  Same issue on Vivaldi, Chrome, Firefox.  I've downloaded the firmware file multiple times.

**To Reproduce**
Steps to reproduce the behavior:
1. Click upload button and select the firmware
2. Screen immediately goes to "Please wait, Applying Settings...", and it never changes.
3. Attempting to go to router home page in another window or browser doesn't work for a few minutes, though the internet connection doesn't go down.  After a few minutes. I can login to the router again, but the firmware hasn't updated.

manishie, 2024-01-10:
I tried with RT-AX95Q_3004_388.4_0-gnuton1_puresqubi.w.  Same issue.

dacootmeister, 2024-01-10:
Question; do you have a USB stick placed ? If so remove it, and repeat the flash process.
I encounter the same with my XT8 and it only flashes a new version when I detach USB..

manishie, 2024-01-10:
@dacootmeister, nope, no USB plugged in, but thanks for the suggestion.

LarryRosen108, 2024-01-10:
Do you have a phone or tablet you can try it from? I just updated my 2 node
XT8 that way with no issues.

Larry
------------------------------------------
Sent from my mobile device. Please excuse spelling and brevity.

gnuton, 2024-01-10:
If you cannot flash your router is because you have a V2. Please check the
he version.

manishie, 2024-01-10:
> If you cannot flash your router is because you have a V2. Please check the he version.
> […](#)

ahhhh, yes indeed I have a v2.  I didn't see the version mentioned on the "supported routers" page, but I just now saw the thread mentioning that only v1 is supported.  thx!

closed 2024-01-10 by manishie