gnuton/asuswrt-merlin.ng #522: TUF-AX5400 2.4 ghz network not working after 7 days

ultravy, 2024-01-11:
**Router Model Affected**
Models: TUF-AX5400 v1

**Firmware Version Affected**
3004.388.5_0-gnuton1
3004.388.4_0-gnuton1
388.2_2_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
No
 
Stable: 388.1_0-gnuton1 - this is the stable version so far, 2.4 working without problems.

**Describe the bug**
After 7 days, the router is kicking 2.4 ghz clients off. Security camera, wifi switches , basically everything running on 2.4 can't connect anymore. 5 Ghz band and ethernet port working without problems . I made 2 separate networks for 2.4 and 5.

Thu, Jan 11 00:46:54 2024* Daylight saving time is enabled in this time zone.
--
7 days 21 hour(s) 0 minute(s) 1 seconds

This a part of a log and is keep going in a loop.

Jan 11 00:50:25 wlceventd: wlceventd_proc_event(645): eth5: Deauth_ind B4:B0:24:55:B8:78, status: 0, reason: Unspecified reason (1), rssi:0
Jan 11 00:50:25 wlceventd: wlceventd_proc_event(685): eth5: Auth B4:B0:24:55:B8:78, status: Successful (0), rssi:0
Jan 11 00:50:25 wlceventd: wlceventd_proc_event(722): eth5: Assoc B4:B0:24:55:B8:78, status: Successful (0), rssi:-43

Second issue is that when i enable 160 mhz channel bandwidth, it's not sitting on my selected channel (100 and above). It always selects whatever channel like is running in auto channel select. Eventually 160 mhz will work but only for couple of hours, falls back to 80 mhz , it happens at random.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash described above firmware 
2. Let it run for 7 days without reboot
3. After 7 days 2.4 ghz network is not working, getting death errors in log 

**Expected behavior**
Do not kick 2.4 ghz clients after 7 days of running. 
Keep selected channel and channel bandwidth : ex c 100 and 160 mhz channel bandwidth.

572289049, 2024-01-11:
[image]
You can fix it like this first and wait for subsequent author updates.

ZenTauren, 2024-02-01:
So this is why I've been having problems with my 3D printer.

ultravy, 2024-02-14:
> ![Snipaste_2024-01-11_12-03-00](https://private-user-images.githubusercontent.com/56331568/295786249-1f00c883-f24d-4f04-9a9a-a02ab36b6ea6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc4NzIyNjYsIm5iZiI6MTcwNzg3MTk2NiwicGF0aCI6Ii81NjMzMTU2OC8yOTU3ODYyNDktMWYwMGM4ODMtZjI0ZC00ZjA0LTlhOWEtYTAyYWIzNmI2ZWE2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE0VDAwNTI0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThiNDA1NGNjNjVkOWU3OTQyYWE5MDEzNjhjYTg5MWFjODlmMzFhNDlkMTJkOTg0MDM4N2RiYTMwYzc5ODFhNTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KFWnITG1bQXR1jSnMFwiutMFuldoxaehfZSJ3btOZeQ) You can fix it like this first and wait for subsequent author updates.

> Scheduled reboot os one way to do it. I also used nightly reboot at 4am to work around issues
> 
> I actually run a new flash of Merlin 388.5 on Asus AX86UPro and experienced something similar but not quite the same, in my case related to wifi on 5G band actually, terrible slowness after a few (2-4?) days, goes away after router reboot. Need to test it more, for now without a scheduled reboot again to see if it ever comes back again.

Yes but after reboot its changing channel from 100 or above is going on 36. it has more power on channel 100 or above.

ZenTauren, 2024-11-30:
This issue still exists in the latest update, annoying to have to reboot the router every week when many devices have to be rebooted as well.