gnuton/asuswrt-merlin.ng #523: 🔥 New Models to support for 2024 🔥

gnuton, 2024-01-12:
Please add here the ONLY the model you wanna see being supported in 2024.

WARNING: People breaking the rules will be BLOCKED.
For chatting use the chat.

gnuton, 2024-01-12:
TUF-AX3000 V2

gnuton, 2024-01-12:
RT-AX58 V2

R3D-PLAN3T, 2024-01-12:
GT-BE98

R3D-PLAN3T, 2024-01-12:
RT-BE96U

R3D-PLAN3T, 2024-01-12:
GT-BE98 Pro

gnuton, 2024-01-12:
XT9 (XT8PRO)

gnuton, 2024-01-12:
ET12

105PM, 2024-01-12:
TUF-AX5400 V2

bartlomiejborzucki, 2024-01-12:
XD6

buciutaandrei, 2024-01-12:
RT-AX59U

calini, 2024-01-12:
Please add support for the RT-AXE7800. Willing to test it!

MWYang, 2024-01-12:
GS-AX3000

ehb224, 2024-01-12:
ZenWiFi-AX-XT8 V2/RT-AX95Q V2

Naolador, 2024-01-13:
RT-AX3000 V2 please. It should be very similar to TUF-AX3000 V2

joelcollet, 2024-01-13:
RT-AX53U

LordPhoenix-ZA, 2024-01-13:
ZenWifi XD4 Mini 

it’s not EOL. Source code is also available from ASUS

AByakugans, 2024-01-14:
ASUS RT-AXE7800 Firmware version 3.0.0.4.388.22068

https://dlcdnets.asus.com/pub/ASUS/wireless/RT-AXE7800/FW_RT_AXE7800_300438822068.zip?model=RT-AXE7800

Please, add support for ASUS RT-AXE7800. It would be awesome. I can test anything you want in that router. Thank you!

aqvarious, 2024-01-15:
Thanks for adding  RT-AXE7800 in the Wishlist for 2024. 

Any idea when that will happen and how would we know that it’s ready !

reaps101, 2024-01-16:
TUF-AX5400

Sirleeofroy, 2024-01-19:
XT9

Bandes, 2024-01-23:
Zenwifi XT9 please

yarosman, 2024-01-24:
ZenWiFi AX Mini

CodeFoxy-Github, 2024-01-25:
RT-AC1300UHP

karrui, 2024-01-27:
ROG Rapture GT6

RikshaDriver, 2024-01-29:
RT-AX1800S (Same MTK chipset as RT-AX53U and RT-AX1800U)

sadi9166, 2024-01-29:
DSL-AC51

Cogginz420, 2024-02-02:
ZenWiFi Hybrid AX (XP4)

kirkrehn83, 2024-02-12:
AC 2600 blue cave

eejjay, 2024-02-12:
GS-AX5400

realizm, 2024-02-13:
RT-AX57

RohhMooHyun, 2024-02-17:
tx ax6000

PaulPizzey247, 2024-02-22:
ZenWiFi XT9

mxsenay, 2024-02-24:
DSL-AC88U

computersteve, 2024-02-26:
ZenWiFi XT9

MauroAsgan, 2024-02-29:
ASUS DSL-AC56U is a must. There are many users of this users, it is essential at least with older Merlin versions.

KyounghoonKwak, 2024-03-07:
TUF-AX5400 V2

kwgrg, 2024-03-10:
TUF-AX4200Q

xczsonn, 2024-03-13:
tx ax3000

ryanpun1223, 2024-03-19:
Zenwifi XD6S

runningbird2, 2024-03-20:
RT-AXE7800

osterac, 2024-03-23:
> RT-AXE7800

I own this router and would be willing to beta test

Panchajanya1999, 2024-04-04:
> RT-AX53U

Didn't he said it's broadcom only???

bitlisz1, 2024-04-06:
ZenWiFi XD6 Series (XD6/XD6S)

O691385, 2024-04-12:
GT-AC5300 
Please support this model. It is a flagship one with cheap price (in South East Asia)
It has same chipset as GT-AC2900, so hopefully it won't take a lot of your time.

FrozenChen7232, 2024-04-24:
RT-BE88U

lazosman, 2024-04-27:
DSL-AC87VG ?

motto77, 2024-04-28:
RT-AXE7800

davidpalmeruk, 2024-05-03:
> ZenWifi XT9 (XT8PRO)

@gnuton any hope for the XT9, as the most voted for model on the list?

testermania, 2024-05-08:
dsl-n16

gps is here @gnuton 

https://github.com/executabls/dsl-n16_gpl

nara1111, 2024-05-08:
ZenWiFi XD6

durip, 2024-05-16:
ZenWiFi XT8 V2 (RT-AX95Q V2)

Niklasg85, 2024-05-21:
> Kindly contribute only the model you wish to see supported in 2024 in a single comment. If your chosen model is already mentioned, please give it a thumbs up. The more thumbs up a model receives, the higher the likelihood of its support.
> 
> ⚠️ CAUTION:
> 
> * Violators of the rules will be blocked and the comment deleted. Use the chat for casual conversations.
> * Banned user cannot interact with this project anymore (creating issues, comments).
> * If you get banned you can contact me via [chat (support channel)](https://discord.com/channels/1155054060848807976/1168099691477614642)
> 
> 🔣 RULES
> 
> * 1 model per comment
> * the comment must have only the model name.
> * thumb up your own comment to give it a vote
> * thumb up the comment of the model if it already exist in this page
> * do not thumb down comments
> * no chats, use [discord](https://discord.com/invite/pm7q6KFSzp).
> * only broadcom based routers can be supported
> * no legacy routers (eg: AC)
> 
> 🔥 LIST OF SELECTED 2024 ROUTERS 🔥 These routers will have continuos support
> 
> * RT-AX5400 - [🌟 Support for RT-AX5400 🌟 #503](https://github.com/gnuton/asuswrt-merlin.ng/issues/503)
> * TUF-AX3000 V2 - [🌟 Support for TUF-AX3000 V2 🌟 #529](https://github.com/gnuton/asuswrt-merlin.ng/issues/529)
> * RT- AX58U V2 - [🌟 Support for RT-AX58U V2 🌟 #550](https://github.com/gnuton/asuswrt-merlin.ng/issues/550)
> 
> 🌟 UNDER DEVELOPMENT 🌟 Models which are under development. Once considered stable and having a good user base it will be selected. The code relative to these models will most probably have same life cycle as the "selected" models.
> 
> * None
> 
> 💖 CONFIRMED POSSIBLE CANDIDATES 💖 These are routers that I already own and that ASUS has already offered its support
> 
> * ROG Rapture GT6
> * RT-AXE7800
> * GT-BE98
> 
> 👎 WON'T SUPPORT 👎 These routes are reaching EOL and I cannot get support from ASUS. Routers in this list which won't be supported will get a thumb down from me.
> 
> * RT-AX55
> * TUF-AX6000
> * DSL-AC88U
> * RT-ACxxx

Love that you do support the TUF-AX3000 V2. Iam using the latest beta1 for the router and it works flawless

davnozdu, 2024-06-04:
GT-BE98

RabbitJun, 2024-06-08:
TUF-BE3600

SuperACc2024, 2024-06-23:
Best connect
DSL AX82U and Aimesh RT 53U
RT 53U Best Price performance

SuperACc2024, 2024-06-23:
RT-AX53U

hyperknot, 2024-06-28:
Hi, AX57 seems to be the same as AX58 v2, with half the RAM and Flash size. Do you think you can add support for it?

AByakugans, 2024-07-08:
Hello,
model RT-AXE7800 is among candidates for a long time it is half year 2024 will you add support for this model, please? I am thinking about buying other ASUS device if this one is not supported by you so to make my final decision... would be awesome to add that. Thank you!

Daxxx1, 2024-07-17:
GS-AX 5400

Disionas, 2024-07-19:
DSL-N14U

[88 more replies]