gnuton/asuswrt-merlin.ng #528: Bug/Feature? USB not recognised by mesh router

johnmhedge, 2024-01-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
My brand new Sandisk not recognised

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I'd like to be able to use the Sandisk as a network repository

**Additional context**
Add any other context or screenshots about the feature request here.
ASUS RT-AX92U Mesh router

gnuton, 2024-01-21:
Hi John,
if your USB drive is not recognized ther are different options.
There are very very few chance this is a softare issue.

If you can provide some logs we can see what's the problem. Usually if it's an external hard drive, you need to provide more power (eg using a powereed USB hub).

johnmhedge, 2024-01-22:
Hi Antonio,
Thanks for the prompt response.
Maybe I'm showing my technology illiteracy. I'd just assumed plugging in a
USB stick was all that was required. They are certainly not powered.
Regards,
John

closed 2024-01-26 by gnuton