gnuton/asuswrt-merlin.ng #529: 🌟 Support for TUF-AX3000 V2 🌟

gnuton, 2024-01-22:
*This thread is about supporting TUF-AX3000 V2.
If you see this issue here is because I have decided to support this model.

So, feel free to ask questions here about this topic.
I will keep the status updated.

Status:
- [x] received GPL
- [x] received development unit
- [ ] Merged GPL to feature-branch
- [ ] Initial testing & Bugfixing phase
- [ ] Public pre-alphas
- [ ] Alpha released
- [ ] collecting testing reports  & Bugfixing phase
- [ ] Beta released
- [ ] collecting testing reports  & Bugfixing phase
- [ ] Stable released

gnuton, 2024-01-22:
Looking for pre-alpha tester. 
Pre-alpha builds may soft-brick your routers. So please be sure you know how to unbrick before applying as alpha tester.
If interested ping me.

Rovasteen, 2024-01-23:
Count me in!

Rovasteen, 2024-01-23:
@gnuton I would like to test the TUF-AX3000 V2 firmware.   Please let me know where to get the firmware..  Thank you!

gnuton, 2024-01-23:
I will publish it as soon as I do have a pre alpha

l0r3n75, 2024-01-25:
I'm willing to also test the pre alpha once you have it ready!

gnuton, 2024-01-31:
I am gonna release the pre-alpha 1 volotunteer required.. 
ping me via discord. 
NOTE: Pre-alpha can scft brick the device.. so please contact me only if you know how to recover.

Willemv2, 2024-01-31:
Can try it out for you i know how to recover, didn't got as far as you are now.

gnuton, 2024-01-31:
pre alpha images:

- [TUF-AX3000_V2 (Default UI)](https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/7729763077/artifacts/1209830970)
- [TUF-AX3000_V2 (TUF UI)](https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/7729763077/artifacts/1209830973)

Note - you must be logged for downlaoding these images.
These are pre-alpha and they can soft brick your router. please do not flash them unless you know how to recover.

Rovasteen, 2024-01-31:
Neither build worked for me.  Flashed the default UI firmware and would not boot up even after setting the IP address manually.  Flashed the 2nd build with TUF UI from recovery mode.  Also did not work. So will go back into recovery mode and flash a stock image to start over again.

gnuton, 2024-02-01:
@Rovasteen thanks for testing. 

It seems like it's soldering time...
(picture from the internet... JTAG port seems to be on the right side)
[image]

Rovasteen, 2024-02-01:
OK.  I can get the recovery console up.  Why can't I flash from there?

Rovasteen, 2024-02-01:
I can easily get the router into recovery mode.  The Asus firmware restoration tool stops at around 79% uploading.  Flashing from the recovery console directly also does not seem to flash an image.  Seems almost like there is an image size limit on the recovery console.   I have never seem a Broadcom router that you could not recover from the recovery console until this one.   Ideas?

Willemv2, 2024-02-01:
> I can easily get the router into recovery mode.  The Asus firmware restoration tool stops at around 79% uploading.  Flashing from the recovery console directly also does not seem to flash an image. [...]

Did u try a lower version of the stock firmware ?

Rovasteen, 2024-02-01:
I tried multiple FW versions including Asus initial image.  I can get into recovery mode easily but none of  the images flash.  @gnuton what do you suggest as the router is NOT hard bricked yet I cannot load an image either.

gnuton, 2024-02-01:
I soft bricked mine too. I have to check what to do. As suggested you should try to flash a old version of the firmware. Are you trying it through ftp or the Asus recovery tool? Just wondering if you are using the right procedure.

Willemv2, 2024-02-01:
> I soft bricked mine too. I have to check what to do. As suggested you should try to flash an old version of the stock firmware. Are you trying it through ftp or the Asus recovery tool? Just wondering if you are using the right procedure.

Indeed first try the older firmware and try ftp side to see if it works or not before i brick mine too😅

Rovasteen, 2024-02-01:
I held down power button and the reset button to get into recovery mode. I set the ethernet port to a fixed IP address as DHCP is not working of course on the router in this state.  Tried flashing from Asus firmware recovery tool and also tried flashing directly from the recovery console using a browser.  Have not tried FTP.   I have a long time ago used TFTP and ping.  What is the FTP procedure?

Rovasteen, 2024-02-01:
I can ping the router all day long and I get back responses:

C:\Windows\System32>ping -t 192.168.1.1

Pinging 192.168.1.1 with 32 bytes of data:
Reply from 192.168.1.1: bytes=32 time<1ms TTL=128
Reply from 192.168.1.1: bytes=32 time<1ms TTL=128
Reply from 192.168.1.1: bytes=32 time<1ms TTL=128
Reply from 192.168.1.1: bytes=32 time<1ms TTL=128
Reply from 192.168.1.1: bytes=32 time<1ms TTL=128
Reply from 192.168.1.1: bytes=32 time<1ms TTL=128

TFTP by command has not worked yet

C:\Windows\System32> tftp -i 192.168.1.1 put C:\TUF-AX3000_V2.pkgtb
 Connect request failed

gnuton, 2024-02-01:
This is the way I do use tftp on linux. If the mode is automatically set to binary than it should be fine.
I still need to test it on my TUF-AX3000 V2 and recoverit.. btw..

#  tftp
tftp> connect 192.168.1.1
tftp> mode binary
tftp> put THE_IMAGE
then wait until ou get that all data has been sent and wait a few minutes 

If you attach a JTAG cable you should see some logs like these
enter tftp mode:
Reading ::
TFTP Server.
..tftp retry wait 0
Failed.: Timeout occured
Reading ::
TFTP Server.
..tftp retry wait 0
Failed.: Timeout occured
Reading ::
TFTP Server.
..tftp retry wait 0
Failed.: Timeout occured
Reading ::

gnuton, 2024-02-01:
Tested and the good news I get same stuff as you @Rovasteen !
tftp> connect 192.168.1.1
tftp> mode binary
tftp> put TUF-AX3000_V2_3.0.0.4_388_23785-g6bb7c92_nand_squashfs.pkgtb
Transfer timed out.

gnuton, 2024-02-02:
@Rovasteen It looks like HND devices are harder to recovery than older and you need to try many many times.
Sometimes yu have to find the right eth/wan port to get it working too. Typically, pinging to a device at CFE would return a TTL of 100 hops instead of 64 hops.

I have done just few tries, but I would next get a [serial adapter](https://openwrt.org/docs/techref/hardware/port.serial.cables) for debuging and recover. Among the sevreal options the serial  CP2102, seems to be the best.

Rovasteen, 2024-02-02:
@gnuton could you contact Asus and get their recommendations as some of the new devices accept to boot from the serial port as an example. Maybe this device has some means of recovery that Asus can turn us onto.

Rovasteen, 2024-02-02:
I can report my router is recovered.   Here is what I did,, which I do not know if it is repeatable.

1) Connect Ethernet to Port 1 of the router,  Set a fixed IP address at your computer as DHCP does not function in recovery mode.
2) Hold down WPS button for 15 seconds and turn on power to do a hard reset.
3) After hard reset, hold down reset button for 15 seconds and power on to put router in recovery mode.
4) Use browser and access the recovery console at 192.168.1.1 or whatever IP address you are using at your computer.  Upload image from the recovery console.  This may take a while.

I used the OEM image TUF-AX3000_V2_3.0.0.4_388_23785-g6bb7c92_nand_squashfs.pkgtb

Rovasteen, 2024-02-02:
I don't know if this router has 2 partitions or not.   It seems like holding down the WPS button during power on may have been what solved this.

gnuton, 2024-02-02:
@Rovasteen yep I managed to recover mine too using the same method.
[image]

gnuton, 2024-02-03:
Build issues found:
````
024-01-31T18:21:03.2144953Z make -C sdparm-1.02
2024-01-31T18:21:03.2225565Z make[5]: Entering directory '/project/asuswrt-merlin.ng/release/src/router/sdparm-1.02'
2024-01-31T18:21:03.2226580Z cd . && /bin/sh /project/asuswrt-merlin.ng/release/src/router/sdparm-1.02/missing --run aclocal-1.10 
2024-01-31T18:21:03.2294851Z /project/asuswrt-merlin.ng/release/src/router/sdparm-1.02/missing: line 52: aclocal-1.10: command not found
2024-01-31T18:21:03.2309572Z WARNING: `aclocal-1.10' is missing on your system.  You should only need it if
2024-01-31T18:21:03.2310824Z          you modified `acinclude.m4' or `configure.ac'.  You might want
2024-01-31T18:21:03.2311925Z          to install the `Automake' and `Perl' packages.  Grab them from
2024-01-31T18:21:03.2312729Z          any GNU archive site.
2024-01-31T18:21:03.2367578Z  cd . && /bin/sh /project/asuswrt-merlin.ng/release/src/router/sdparm-1.02/missing --run automake-1.10 --gnu 
2024-01-31T18:21:03.2411228Z /project/asuswrt-merlin.ng/release/src/router/sdparm-1.02/missing: line 52: automake-1.10: command not found
2024-01-31T18:21:03.2424682Z WARNING: `automake-1.10' is missing on your system.  You should only need it if
2024-01-31T18:21:03.2425547Z          you modified `Makefile.am', `acinclude.m4' or `configure.ac'.
2024-01-31T18:21:03.2426201Z          You might want to install the `Automake' and `Perl' packages.
2024-01-31T18:21:03.2426723Z          Grab them from any GNU archive site.
2024-01-31T18:21:03.2543636Z cd . && /bin/sh /project/asuswrt-merlin.ng/release/src/router/sdparm-1.02/missing --run autoconf
2024-01-31T18:21:03.3558579Z aclocal.m4:14: error: this file was generated for autoconf 2.61.
2024-01-31T18:21:03.3559621Z You have another version of autoconf.  If you want to use that,
2024-01-31T18:21:03.3560205Z you should regenerate the build system entirely.
2024-01-31T18:21:03.3560626Z aclocal.m4:14: the top level
2024-01-31T18:21:03.3564239Z autom4te: /usr/bin/m4 failed with exit status: 63
2024-01-31T18:21:03.3588974Z WARNING: `autoconf' is probably too old.  You should only need it if
2024-01-31T18:21:03.3590407Z          you modified `configure.ac'.  You might want to install the
2024-01-31T18:21:03.3591216Z          `Autoconf' and `GNU m4' packages.  Grab them from any GNU
2024-01-31T18:21:03.3591724Z          archive site.
```

Rovasteen, 2024-02-03:
Awesome progress @gnuton !  Looking forward to the first bootable build.  Hopefully, Asus is also providing latest Broadcom drivers (at least I can hope for that).  Take care and thank you for supporting this router!

Rovasteen, 2024-02-09:
@gnuton you've done a really professional job setting up your github.  Everything is extremely well organized and well labeled.  It's been interesting looking at Actions to see how the test builds are coming along.  Many thanks to you!

gnuton, 2024-02-09:
Thanks. Honestly I am not happy about it. The setup on circle I was better. There are still issues with permissions and some bugs in some actions which are really annoying. In any case the setup works. But it's not final.

Rovasteen, 2024-02-12:
@gnuton does the build generated successfully on 02/06 load?  Ok to try that one?

Rovasteen, 2024-02-20:
@gnuton you may know already but M_ars has a project here:  https://bitbucket.org/M_ars/freshtomato-ax/commits/  to port the TUF-AX3000 v2 to Tomato.  The project has been able to compile builds and introduce some updates but has not ported a build to the Tomato interface.  Anyway, maybe there are some learnings to be gleaned from another router aficionado.

gnuton, 2024-02-20:
> @gnuton you may know already but M_ars has a project here:  https://bitbucket.org/M_ars/freshtomato-ax/commits/  to port the TUF-AX3000 v2 to Tomato.  The project has been able to compile builds and [...]

Thanks I am aware of this port. The problem I am facing needs debugging and some time for me to solder stuff on the board.. I just need to find some spare time for this.

Rovasteen, 2024-02-20:
I had no idea you were doing debugging at board level.

Rovasteen, 2024-02-26:
@gnuton do the ITUF-AX3000 v2 mages #216: Commit aef4298 pushed by gnuton on 02/25 load so we can try them out?

Rovasteen, 2024-02-26:
OK.  Cool.  Will look at what you sent me.   Hopefully Asus is supporting this process.

gnuton, 2024-02-28:
@Rovasteen the SDK comes with the GPL. These are standard issues I get when merging code into Merlin.
I am not stuck, I just need the time to get to solder a few jumpers and get proper debugging messages to understand what's the problem.

gnuton, 2024-03-03:
boot log https://gist.github.com/gnuton/69a417ea7ab9be31dde188321c7e5c22

Rovasteen, 2024-03-04:
So I guess you wired up the serial port.  Looking good!  That is progress.  I am curious (and interested) about the fault.  What is the fault that you see?

gnuton, 2024-03-08:
@Rovasteen Issues raising the network interfaces. The problem was the archer driver, You can find errors related to it in the logs.

gnuton, 2024-03-08:
pre-alpha images 
default
https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8178170034/artifacts/1303732601
[image]

tuf
https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8178170034/artifacts/1303732602
[image]

Rovasteen, 2024-03-08:
Running on your build!  Congrats.  

Working:   WAN 1Gbps, WIFI 2.4, WIFI 5.0, IPv6, DNS over TLS.  

Issues:  

5 GHz radio signal strength seems a little weak.  Not sure if there are any tweaks or not, but definitely weaker than TUF-AX5400 that I have.  My PC is 2x2 using Intel AX210.  With 2 other routers I can connect at 1.1 Gbps, with the TUF-AX3000 v2 I am connecting at 649 Mbps.  This is due to the low signal strength.  Using Channel 36 and same settings as other routers.

Tools -> CPU Frequency displays "??? MHz".   

LAN -> IP Address default is 192.168.50.1 which is the Asus default, are you using that or the normal Merlin default of 192.168.1.1 ?

gnuton, 2024-03-08:
Tools -> CPU Frequency displays "??? MHz". - known issue. The cPU frequency is hardcoded and not detected.

LAN for me is 
~$ host router.asus.com
router.asus.com has address 192.168.51.1 the asus one.

As for th 5Ghz signal: there is not so much I can do as far as I know. usually those are regulated from the settings you give to the driver, more than the driver itself which should be the latest.

Rovasteen, 2024-03-14:
These builds seem stable.  Will try NVRAM settings to improve wireless performance as I guess these days you can't flash the CFE any longer.

Rovasteen, 2024-03-18:
I do think that the 5Ghz radio signal seems very weak.  Sorry for being insistent.  I am not saying this is caused by the Gnuton build as the old factory Asus build behaves the same way.

gnuton, 2024-03-19:
@Rovasteen Are you comparing the signal between different firmware images for the same router and unit or what?
how do you come to this?

Rovasteen, 2024-03-19:
I believe the issue is that the TUF-AX3000 v2 is using EU country code settings while my other routers are using US.  Checked via a dump of NVRAM.   It used to be not long ago that CFE could be edited but guess that is not possible any longer. So I guess the best one can do is to set parameters for the country and put that in a script file?

gnuton, 2024-03-20:
I guess you would need to set the parameters just once , then commit the nvram. If that is overridden at not, the a script is required. But changing country would violate local rules. So not recommended.

Rovasteen, 2024-03-22:
A shout out for @gnuton for making this possible.  Many thanks to @gnuton for going the extra kilometer to prepare this firmware.  I know that it is a ton of work to prepare the environment, debug buggy firmware and compile.  Thank you!

YoshiKill, 2024-03-23:
Any date or time when the beta is coming? Maybe simple users like (me)us can somehow help?

gnuton, 2024-03-24:
New alpha
https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8403778904/artifacts/1352759152 - default
https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8403778904/artifacts/1352759154 - tuf

Tools page should display frequency
NOTE: untested images so potential bricks may happen.

SMOL25, 2024-03-24:
Hello. What is the file to download? AX3000 and AX 3000v2 have the same name at the release page.

Willemv2, 2024-03-24:
> Hello. What is the file to download? AX3000 and AX 3000v2 have the same name at the release page.

Hi this page is specificaly for the AX3000-V2 if you are looking for the first gen one please look at the main page where you can find the link for the AX-3000 software.

Willemv2, 2024-03-24:
> Hello. What is the file to download? AX3000 and AX 3000v2 have the same name at the release page.

The first AX-3000 router uses the same software as the AX-58U router!

SMOL25, 2024-03-24:
> > Hello. What is the file to download? AX3000 and AX 3000v2 have the same name at the release page.
> 
> Hi this page is specificaly for the AX3000-V2 if you are looking for the first gen one please look at the main page where you can find the link for the AX-3000 software. 

No, I want download firmware for TUF ax3000v2 but don't know what is...

Willemv2, 2024-03-24:
> No, I want download firmware for TUF ax3000v2 but don't know what is...

For now these are the newest test images. But keep in mind these are still in alpha stage of release and can brick your router, so use with caution! Please only use with knowledge and keep everyone updated on this page what your experience is. 
Here is the link to the image: 

New alpha
https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8403778904/artifacts/1352759152 - default
https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8403778904/artifacts/1352759154 - tuf

Tools page should display frequency
NOTE: untested images so potential bricks may happen.

SMOL25, 2024-03-24:
I thought it was stable image. I will wait for the stable image. Thanks

johncrichton, 2024-03-24:
hey @gnuton I've been following requests for the ax3000 v2 and when I saw that you are working on supporting it I was really happy. Although I'm not inclined to test pre-alpha versions that might brick my main, only unit that I use, I hope that a small donation will show my appreciation for your work :+1:

mycek91, 2024-03-25:
Excellent work @gnuton, thank you for your hard work and making this build possible!
I also would like to report a bug.
NFS exports don't work:
Similar to this issue #501

gnuton, 2024-03-25:
@mycek91 do you get any any errore in the JavaScript console of your browser? Any log?

DavidGaribov, 2024-03-26:
Hello, dear gnuton! Thank you for your work! After installing the alpha version, the amount of occupied RAM decreased significantly (from 92-96% on stock to 60-71% on your firmware). Tell me, can I post a link to your firmware in the topic of our router on the 4pda website?

[34 more replies]

closed 2024-04-06 by gnuton