gnuton/asuswrt-merlin.ng #530: RT-AX92U reports high load average regardless of system load

pixdrift, 2024-01-24:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
Linux RT-AX92U-0480 4.1.51 #2 SMP PREEMPT Fri Dec 29 23:56:03 UTC 2023 aarch64 ASUSWRT-Merlin

**Is this bug present in upstream Merlin releases too?**
Unknown, will try and find references to similar on other systems.

**Describe the bug**
Regardless of load on the system, the system load metrics consistently report close to full resource utilisation. Running on an RT-AX92U with 2 cores, the load average remains consistently at ~1.9 when the system is completely idle.

```
Mem: 246856K used, 161996K free, 4864K shrd, 0K buff, 27380K cached
CPU0:  0.2% usr  0.8% sys  0.0% nic 98.5% idle  0.0% io  0.0% irq  0.4% sirq
CPU1:  0.2% usr  1.0% sys  0.0% nic 98.7% idle  0.0% io  0.0% irq  0.0% sirq
Load average: 1.84 1.90 1.95 1/149 22582
```

There appears to be no discernible contributor to these load numbers, cpu usage, io wait, interrupts etc. 

Note: This node is currently in an AiMesh configuration, unsure if this is contributing.

**To Reproduce**
Steps to reproduce the behaviour:
1. SSH to RT-AX92U-0480 (stock after flash)
2. Execute 'top' and press '1' to breakdown to single CPU view and compare running processes/performance with load average displayed
3. Alternatively, execute 'uptime' to view load average, or it can be viewed from the Web UI

**Expected behaviour**
Load average should be representative of the system's load. Given the base load on an idle RT-AX92U, my expectations would be an idle load average of <0.5.

**Screenshots**
N/A console posted above

pixdrift, 2024-01-24:
Similar issues noted in this kernel issue, but versions aren't quite the same (unless problematic patch was backported?).
https://github.com/raspberrypi/linux/issues/2881

gnuton, 2024-01-26:
"On Unix-like systems, including Linux, the system load is a measurement of the computational work the system is performing. This measurement is displayed as a number. A completely idle computer has a load average of 0. Each running process either using or waiting for CPU resources adds 1 to the load average. So, if your system has a load of 5, five processes are either using or waiting for the CPU."

from https://www.howtogeek.com/194642/understanding-the-load-average-on-linux-and-other-unix-like-systems/

I would not be worried about this, this is the way the linux kernel collects statistics. There may be a better way to calculate them, but as far as I know how to interpret it, I think it's not an issue.

pixdrift, 2024-01-26:
Thanks @gnuton.

 I understand what system load is, and also what the numbers are meant to represent. The reason for this ticket is that this build is not displaying accurate information as described in the article you linked. 
 
 A system load of ~1.9 on an idle system with 2 cores is incorrect. From the ps output there don't appear to be any processes in 'R' (runnable) or 'D' (uninterruptible sleep) states to contribute to the load.
 
 I accept that this is unlikely an issue specific to this fork. but may possibly be a kernel driver/configuration issue.

gnuton, 2024-01-26:
Reopened and ready to get a patch for this.