gnuton/asuswrt-merlin.ng #531: Crash restart on 3004_388.5_0 and 3004_388.4_0

craigh1960, 2024-01-25:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**

3004_388.5_0
3004_388.4_0

**Is this bug present in upstream Merlin releases too?**
Yes/No 

No idea sorry

**Describe the bug**

I have been running Version : 388.2_2-gnuton1 with no issues for a number of months on my DSL-AX82U in Australia On FTTN.

Decided to upgrade to 3004_388.5_0 and all seemed good for a number of days then I had a crash and after the crash restart it would not connect at a good speed. After a hard reset it was good for a number of days again then crash.

went back to 388.2 and stable fo extended period

Tried 3004_388.4_0 and same crash behaviour as 388.5 after a number of days stable operation.

log attached

[syslog.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/14058348/syslog.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

gnuton, 2024-01-26:
Suggestion: reset to factory

craigh1960, 2024-01-31:
I reloaded 3004_388.5_0, performed a factory reset and loaded my config.
Has been running for 4 days without an issue so looking good.  Will post again in a few days to confirm. Thanks

craigh1960, 2024-02-04:
8 days and no crash, this can be closed out. Must have been some corruption with upgrade.

gnuton, 2024-02-05:
BTW if you reset to factory and you reload your config you import all your "mess" again.

closed 2024-02-05 by gnuton

craigh1960, 2024-02-05:
Yes I was aware that could be the case but my first try was with existing config then build it again if the problem continued.  May still do that anyway