gnuton/asuswrt-merlin.ng #532: ZenWiFi XT8 / RT-AX95Q Mesh Nodes not showing correctly

robbbaxley, 2024-01-26:
**Router Model Affected**
Models: Asus ZenWiFi XT8 / RT-AX95Q

**Firmware Version Affected**
RT-AX95Q_3004_388.6_0-gnuton0_beta1_puresqubi

**Is this bug present in upstream Merlin releases too?**
Unknown, but not present in earlier Gnuton releases

**Describe the bug**
Mesh Nodes not showing correctly

**To Reproduce**
Steps to reproduce the behavior:
1. Open Main Page (Notice no Mesh noted)
2.  Click on Network Map>Clients>View List (Nodes show here via IP Address only)

**Expected behavior**
Mesh nodes to be accessible via Network Map>Clients>View List (

**Screenshots**

[image]
[image]

robbbaxley, 2024-01-26:
I should note that I realized later that none of the nodes are actually working now. Not even the XT9's which are running Asus firmware. All nodes are showing 'RED'. I can reach the nodes directly via IP address and this is what I see after loggin-in:

The XT8's:

[image]

The XT9's

[image]

robbbaxley, 2024-01-27:
Further testing: I installed the previous Gnuton Firmware RT-AX95Q_3004_388.5_0-gnuton1_puresqubi on ONLY the primary (master) node which restored connectivity to all remote nodes, both the XT8's and XT9's.

dacootmeister, 2024-01-28:
Hi @robbbaxley I got sort of the same setup as you do, but my Meshnodes are all connected without issue after (dirty)flashing them to 3004.388.6_0-gnuton0_beta1.
Repeat the procedure, starting with the main node?

[image]
[image]

robbbaxley, 2024-01-28:
@dacootmeister , I will try that but in the past, using Gnuton or Asus firmware I have never had this problem before. My process is to update the remote Mesh nodes first and then update the primary/master node last. This has always worked with no issues at all until this firmware version.  I don't know what has possibly changed that would cause this issue but when I have time I will  try your method...but I will need to downgrade the remote Mesh nodes first.

robbbaxley, 2024-02-15:
I loaded the new beta 2, [3004.388.6_0-gnuton0_beta2], and still have the issue as the earlier Beta.

However, now, I am no longer able to downgrade to 3004.388.5_0-gnuton1 like I did for Beta 1 so I am a bit stuck for the moment.

gnuton, 2024-02-15:
@robbbaxley do you see any crash in the logs due to downgrade?

robbbaxley, 2024-02-15:
@gnuton, here is the log from my downgrade attemt this morning which, strangely, worked (3rd attempt):

Feb 15 06:59:09 rc_service: httpd 8840:notify_rc stop_upgrade
Feb 15 06:59:09 usb: USB is ejected
Feb 15 06:59:09 WEBDAV_Server: daemon is stopped
Feb 15 06:59:09 WEBDAV_Server: arpping daemon is stopped
Feb 15 06:59:09 iTunes: daemon is stopped
Feb 15 06:59:09 FTP_Server: daemon is stopped
Feb 15 06:59:09 Samba_Server: smb daemon is stopped
Feb 15 06:59:09 Timemachine: daemon is stopped
Feb 15 06:59:09 WEBDAV_Server: daemon is stopped
Feb 15 06:59:10 iTunes: daemon is stopped
Feb 15 06:59:10 FTP_Server: daemon is stopped
Feb 15 06:59:10 Samba_Server: smb daemon is stopped
Feb 15 06:59:10 Timemachine: daemon is stopped
Feb 15 06:59:11 rc_service: hotplug 27826:notify_rc restart_nasapps
Feb 15 06:59:11 rc_service: waitting "stop_upgrade" via httpd ...
Feb 15 06:59:11 ntpd: Stopped ntpd
Feb 15 06:59:13 ahs: ===Terminate ahs daemon===
Feb 15 06:59:14 disk_monitor: Finish
Feb 15 06:59:14 Mastiff: Got SIGTERM
Feb 15 06:59:18 rc_service: hotplug 27844:notify_rc restart_nasapps
Feb 15 06:59:18 rc_service: waitting "restart_nasapps" via hotplug ...
Feb 15 06:59:18 FTP_Server: daemon is stopped
Feb 15 06:59:18 Samba_Server: smb daemon is stopped
Feb 15 06:59:18 Timemachine: daemon is stopped
Feb 15 06:59:19 Samba_Server: daemon is started
Feb 15 06:59:19 FTP_Server: daemon is stopped
Feb 15 06:59:19 wsdd2[27885]: Terminated received.
Feb 15 06:59:21 Samba_Server: smb daemon is stopped
Feb 15 06:59:21 Timemachine: daemon is stopped
Feb 15 06:59:21 Samba_Server: daemon is started
Feb 15 06:59:32 rc_service: httpd 8840:notify_rc start_upgrade
Feb 15 06:59:56 WEBDAV_Server: daemon is stopped
Feb 15 06:59:56 FTP_Server: daemon is stopped
Feb 15 06:59:56 wsdd2[27961]: Terminated received.
Feb 15 06:59:57 Samba_Server: smb daemon is stopped
Feb 15 06:59:57 Timemachine: daemon is stopped
Feb 15 06:59:58 kernel: Flushing net_device wl0.2.
Feb 15 06:59:58 kernel: CSIMON:  CSIMON[1.1.0] Initialization
Feb 15 06:59:58 kernel: CSIMON: M2M usr already registered ...
Feb 15 06:59:59 kernel: CSIMON:  CSIMON[1.1.0] Initialization
Feb 15 06:59:59 kernel: CSIMON: M2M usr already registered ...
Feb 15 07:00:01 kernel: port_generic_stop 560 skip turnning off power on eth0 here
Feb 15 07:00:01 kernel: port_generic_stop 560 skip turnning off power on eth1 here
Feb 15 07:00:01 kernel: port_generic_stop 560 skip turnning off power on eth2 here
Feb 15 07:00:02 kernel: port_generic_stop 560 skip turnning off power on eth3 here
Feb 15 07:00:02 kernel: ubi2: attaching mtd10
Feb 15 07:00:02 kernel: ubi2: scanning is finished
Feb 15 07:00:02 kernel: ubi2: attached mtd10 (name "misc1", size 8 MiB)
Feb 15 07:00:02 kernel: ubi2: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
Feb 15 07:00:02 kernel: ubi2: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
Feb 15 07:00:02 kernel: ubi2: VID header offset: 2048 (aligned 2048), data offset: 4096
Feb 15 07:00:02 kernel: ubi2: good PEBs: 64, bad PEBs: 0, corrupted PEBs: 0
Feb 15 07:00:02 kernel: ubi2: user volume: 1, internal volumes: 1, max. volumes count: 128
Feb 15 07:00:02 kernel: ubi2: max/mean erase counter: 60/27, WL threshold: 4096, image sequence number: 1526576606
Feb 15 07:00:02 kernel: ubi2: available PEBs: 0, total reserved PEBs: 64, PEBs reserved for bad PEB handling: 4
Feb 15 07:00:02 kernel: ubi2: background thread "ubi_bgt2d" started, PID 28132
Feb 15 07:00:03 kernel: UBIFS (ubi2:0): background thread "ubifs_bgt2_0" started, PID 28142
Feb 15 07:00:03 kernel: UBIFS (ubi2:0): UBIFS: mounted UBI device 2, volume 0, name "nvram"
Feb 15 07:00:03 kernel: UBIFS (ubi2:0): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
Feb 15 07:00:03 kernel: UBIFS (ubi2:0): FS size: 5840896 bytes (5 MiB, 46 LEBs), journal size 1396736 bytes (1 MiB, 11 LEBs)
Feb 15 07:00:03 kernel: UBIFS (ubi2:0): reserved for root: 0 bytes (0 KiB)
Feb 15 07:00:03 kernel: UBIFS (ubi2:0): media format: w4/r0 (latest is w4/r0), UUID 9DC75FF7-E8C0-4018-A1ED-1D54D1E5AC77, small LPT model
Feb 15 07:00:03 kernel: UBIFS (ubi2:0): un-mount UBI device 2
Feb 15 07:00:03 kernel: UBIFS (ubi2:0): background thread "ubifs_bgt2_0" stops
Feb 15 07:00:03 kernel: ubi2: detaching mtd10
Feb 15 07:00:03 kernel: ubi2: mtd10 is detached
Feb 15 07:00:03 FTP_Server: daemon is stopped
Feb 15 07:00:03 Samba_Server: smb daemon is stopped
[... 494 lines cut ...]
Feb 15 07:01:24 avahi-daemon[4158]: Alias name "ZenWiFi_XT8" successfully established.
Feb 15 07:01:25 avahi-daemon[4184]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Feb 15 07:01:25 iTunes_Server: daemon is started
Feb 15 07:01:26 avahi-daemon[4184]: Alias name "ZenWiFi_XT8" successfully established.
Feb 15 07:01:29 crond[1824]: time disparity of 3041695 minutes detected
Feb 15 07:01:33 mt-daapd[4180]: Starting signal handler
Feb 15 07:01:34 mt-daapd[4180]: Loading playlists
Feb 15 07:01:34 mt-daapd[4180]: Initializing database
Feb 15 07:01:34 mt-daapd[4180]: Starting mp3 scan
Feb 15 07:02:52 ahs: [read_json]Update ahs JSON file.
Feb 15 07:03:36 mt-daapd[4180]: Starting web server from /rom/etc/web on port 3689
Feb 15 07:03:36 mt-daapd[4180]: Scanned 6913 songs in  132 seconds
Feb 15 07:03:40 wlceventd: wlceventd_proc_event(645): eth5: Deauth_ind 88:B6:EE:7D:BB:32, status: 0, reason: Disassociated due to inactivity (4), rssi:-60
[... cut ...]
Feb 15 07:03:59 wlceventd: wlceventd_proc_event(662): eth4: Disassoc 10:BF:67:77:D4:87, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Feb 15 07:06:06 rc_service: httpd 2610:notify_rc restart_logger
Feb 15 07:06:06 kernel: klogd started: BusyBox v1.25.1 (2023-12-31 07:18:45 UTC)
Feb 15 07:06:34 rc_service: httpd 2610:notify_rc restart_logger
Feb 15 07:06:34 kernel: klogd started: BusyBox v1.25.1 (2023-12-31 07:18:45 UTC)
Feb 15 07:06:58 kernel: klogd started: BusyBox v1.25.1 (2023-12-31 07:18:45 UTC)

robbbaxley, 2024-02-15:
I have no idea why the downgrade worked this morning on my 3rd attempt but glad it did. Once I was able to successfully load the downgraded firmware, I attempted to upgrade the remote/slave XT8 nodes which did work and, so far, appear to be functioning normally.

[image]

gnuton, 2024-02-15:
I have not checked this issue yet, but usually the errors like the one in ths github issue, are the ones you can fix by cleaning your NVRAM. There may be something bad there,
Another way to debug the problem is to press F12 and check the network call or the js console. if there are errors you get a red box

robbbaxley, 2024-02-16:
@gnuton , quick question for you (or anyone else who knows the answer)...how do you clear the NVRAM on the XT8? I thought I knew how, but...no, I don't remember LOL

gnuton, 2024-02-16:
Reset to factory. Nvram contains the settings

GittyScott, 2024-02-20:
I believe resetting by holding the wps button during boot does this on most Asus routers? 
But if not, you can ssh in and run a command for it 
NB. There's a few variations of the command between versions, you might need to research which is best for your router before running.

gnuton, 2024-03-14:
@robbbaxley I will close this as duplicate of https://github.com/gnuton/asuswrt-merlin.ng/issues/532
In case I am wrong I will reopen this issue again

closed 2024-03-14 by gnuton

robbbaxley, 2024-03-24:
@gnuton I concur with you that this issue is likely related to or a duplicate of #542