gnuton/asuswrt-merlin.ng #535: XT8 - mesh no work wifi

misqkrk, 2024-02-06:
Models: XT8
3004.388.6_0-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
I don't know

2x XT8 - cannot connect to each other (cable connection)
They connect but the wifi doesn't work, it glows blue, in the logs:

Feb  6 13:18:07 bluetoothd[6918]: Bluetooth daemon 5.56
Feb  6 13:18:07 bluetoothd[6918]: src/main.c:main() Unable to get on D-Bus
Feb  6 13:18:08 rc_service: watchdog 1805:notify_rc start_bluetooth_service
Feb  6 13:18:08 rc_service: waitting "start_bluetooth_service" via watchdog ...
Feb  6 13:18:10 BLE: daemon is started
Feb  6 13:18:11 bluetoothd[6957]: Bluetooth daemon 5.56
Feb  6 13:18:11 bluetoothd[6957]: src/main.c:main() Unable to get on D-Bus
Feb  6 13:18:12 rc_service: watchdog 1805:notify_rc start_bluetooth_service
Feb  6 13:18:12 rc_service: waitting "start_bluetooth_service" via watchdog ...
Feb  6 13:18:14 BLE: daemon is started
Feb  6 13:18:15 bluetoothd[7007]: Bluetooth daemon 5.56
Feb  6 13:18:15 bluetoothd[7007]: src/main.c:main() Unable to get on D-Bus
Feb  6 13:18:16 rc_service: watchdog 1805:notify_rc start_bluetooth_service
Feb  6 13:18:16 rc_service: waitting "start_bluetooth_service" via watchdog ...
Feb  6 13:18:18 BLE: daemon is started
Feb  6 13:18:19 bluetoothd[7038]: Bluetooth daemon 5.56
Feb  6 13:18:19 bluetoothd[7038]: src/main.c:main() Unable to get on D-Bus
Feb  6 13:18:20 rc_service: watchdog 1805:notify_rc start_bluetooth_service
Feb  6 13:18:20 rc_service: waitting "start_bluetooth_service" via watchdog ...
Feb  6 13:18:22 BLE: daemon is started
Feb  6 13:18:23 bluetoothd[7077]: Bluetooth daemon 5.56
Feb  6 13:18:23 bluetoothd[7077]: src/main.c:main() Unable to get on D-Bus
Feb  6 13:18:24 rc_service: watchdog 1805:notify_rc start_bluetooth_service
Feb  6 13:18:24 rc_service: waitting "start_bluetooth_service" via watchdog ...

robbbaxley, 2024-02-06:
I wonder if this is related to the issue I reported a few days ago? Issues #532

gnuton, 2024-02-06:
@misqkrk those logs are for the bluetooth.  The bluetooth is used for pairing routers from the same box, but it's not supported.

If led blue is blinking, reset to factory the node and forget the association on the master node. Better if you reset also the master.
Then start the pairing through cable again.
@robbbaxley has big mesh and he has shown that the router connect eachother.
For me this is a support request for now more than a bug. if I am wrong I would be happy to change the status.

gnuton, 2024-02-06:
The bluetooth and the D-Bus can be an issue, but again it's not used.

CPngN, 2024-02-07:
> After uploading the software, it worked for 4 days, but after restarting
> it stopped working - for the "mesh" I have now uploaded the official
> software - I left the software from you on the main one.
> Unfortunately, I didn't have time to restore the main router settings
> today - tomorrow I'm leaving for a 2-month trip to Asia - so I didn't risk
> it anymore.

Oh boy, good luck.  I don't think I'd trust these damn things to run for 2
solid months without intervention.  Although maybe if you aren't home,
actively connecting and using them yourself they'll stay stable.

gnuton, 2024-02-07:
@misqkrk I have a mesh network with some ET8/XT8s all on stock firmware. From time to time the link is lost and I need to link them back. No idea why this happens yet.
BTW just for the record  the AiMesh on stock firmware is the very same than the one on Merlin.. since it's a closed component which we bundle in the firmware.

robbbaxley, 2024-02-08:
I used to see issues with the Mesh (specifically on the 5Ghz network) so I used the built-in reboot scheduler to restart the routers nightly. It's been quite a while since I have had issues like this, but I still schedule the routers to reboot weekly...just in case :)

gnuton, 2024-04-06:
duplicate of https://github.com/gnuton/asuswrt-merlin.ng/issues/535

closed 2024-04-06 by gnuton