gnuton/asuswrt-merlin.ng #541: cannot import my own ssl certificate

mmurza, 2024-02-19:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
- 3004.388.6_0-gnuton0_beta2
- Unstable: 3004.388.6_0-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
I don't know. 
But issue NOT present on:
- Stable: 3004.388.5_0-gnuton1

**Describe the bug**
Cannot import my own SSL certificate for https interface (via "Webui SSL Certificate" section on WAN \ DDNS).
Upload web interface opens, all seems to be OK - I can upload certificate and key files. But this makes no effect - changes are reverted back to auto (localy generated ssl certificate).
Even if there were working custom certificate on previous firmware version (e.g. 388.5 stable), after upgrade to 388.6 beta, web interface reverts to locally generated ssl certificate.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to WAN \ DDNS in web interface
2. Under the section "Webui SSL Certificate", select "Import Your Own Certificate", click upload
3. Using upload web interface, select certificate and key files from your PC. Upload, Apply
4. See https interface - it reverts back to auto (locally generated certificate). Same could be seen on the browser interface (warning for https with untrusted ssl certificate)

Tried renaming certificate and key file names before upload (some forums sugested exact names required: cert.pem and key.pem respectively)
Tried also to make changes via ssh (works on 388.5 stable release for RT-AX92U in mesh configuration, where web interface not available), this doesn't work on this 388.6 beta anymore as well:

create or upload certificate and key files under /tmp/etc
/tmp/etc/cert.pem
/tmp/etc/key.pem

then:
cd /tmp
tar zcvf cert.tgz etc/cert.pem etc/key.pem
mv /jffs/cert.tgz /jffs/cert.tgz.bak
mv cert.tgz /jffs/
service restart_httpd

**Expected behavior**
Selection stays on "Import Your Own Certificate" on web interface under WAN \ DDNS (Webui SSL Certificate), details of imported certificate shown below.

**Screenshots**
[image]

mmurza, 2024-02-19:
My certificate issued by local CA
CN = router.asus.com

Subject alternative names:
DNS Name=router.asus.com
DNS Name=asusrouter.com
DNS Name=router.local
DNS Name=repeater.local
DNS Name=_ownDistinctName_.asuscomm.com
DNS Name=mesh1.local
DNS Name=mesh2.local
DNS Name=mesh3.local
IP Address=192.168._localIP1_
IP Address=192.168._localIP1_
IP Address=192.168._localIP1_
IP Address=192.168._localIP1_
IP Address=192.168._localIP1_

Log entries with word _cert_, while trying to upload custom certificate:

	Line 35: Feb 19 10:44:18 httpd: Delete uploaded certificate
	Line 52: Feb 19 10:44:20 rc_service: httpds 13801:notify_rc restart_ddns;prepare_cert;restart_webdav
	Line 57: Feb 19 10:44:20 custom_script: Running /jffs/scripts/service-event (args:  prepare_cert)
	Line 62: Feb 19 10:44:21 inadyn[21672]: Certificate verification error:num=20:unable to get local issuer certificate:depth=0:/C=TW/ST=Taipei/O=ASUSTeK COMPUTER INC./CN=ns1.asuscomm.com
	Line 63: Feb 19 10:44:21 inadyn[21672]: Certificate verification error:num=21:unable to verify the first certificate:depth=0:/C=TW/ST=Taipei/O=ASUSTeK COMPUTER INC./CN=ns1.asuscomm.com

gnuton, 2024-02-23:
the problem is in the certificate chain verification.

openssl s_client -connect ns1.asuscomm.com:443

should fail. I tested it on beta2 but it works for me.

mmurza, 2024-03-04:
And fun fact - below described method to specify custom certificate on mesh router works.
But not on the main router: after httpd restart, certificate is reverted back to locally speciffied

cd /tmp
mkdir etc
cat << EOF > etc/cert.pem
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
EOF

cat << EOF > etc/key.pem
-----BEGIN RSA PRIVATE KEY-----
...
-----END RSA PRIVATE KEY-----
EOF

tar zcvf cert.tgz etc/cert.pem etc/key.pem
mv /jffs/cert.tgz /jffs/cert.tgz.bak
mv cert.tgz /jffs/
service restart_httpd