gnuton/asuswrt-merlin.ng #542: Asus ZenWiFi Mesh Bug with 388.6 firmware

craigbailey18, 2024-02-20:
**Router Model Affected**
[ASUS ZenWiFi AX (XT8)]

**Firmware Version Affected**
[RT-AX95Q_3004_388.6_0-gnuton0_beta1_puresqubi.w] and [RT-AX95Q_3004_388.6_0-gnuton0_beta2_puresqubi.w]

**Is this bug present in upstream Merlin releases too?**
No, just both RT-AX95Q_3004_388.6 beta 1 and beta 2

**Describe the bug**
The Mesh nodes are not detected on the main router on this firmware. I have the RT-AX95Q and the firmware updates but the nodes don't show up. I have to downgrade the main router back to the older stable version for it to detect them again. All of my 3 nodes are the same model.

**To Reproduce**
1. Update all the nodes to the latest beta firmware
2. Update the main node (router/controller) to the latest firmware
3. You will notice the nodes/mesh points are not detected or connected to the main node either wired or wirelessly anymore.

**Expected behavior**
I usually update it node's firmware first and then the main router. I followed the same process here and the nodes/mesh points never get detected. I even reset all the nodes and started from scratch and the issue remained. I reverted the main router/controller to the stable [3004.388.5_0-gnuton1] and the mesh points were detected again.

**Screenshots**
RT-AX95Q_3004_388.6 Beta 2 Screenshot
[image]

RT-AX95Q_3004_388.5 Stable Screenshot
[image]
[image]

robbbaxley, 2024-02-20:
This is the exact same issue I am seeing on my XT8's and I have 5 nodes (primary + 4 secondary mesh). 

All of my mesh nodes are wired, are yours wired or Wi-Fi backhaul?

craigbailey18, 2024-02-21:
Yes my 2 mesh nodes are on wired backhaul.

robbbaxley, 2024-02-21:
Hm, that's also just like me. My issue #532 is exactly the same as yours. This weekend I will try the 'fix' @gnuton suggested which is to clean the nvram then try the firmware again. Hopefully that will work, but I have always been successful with a 'normal' dirty flash.

gnuton, 2024-02-21:
If you press f12 do you see any error in the developer console or chrome?

craigbailey18, 2024-02-21:
@robbbaxley I will have to probably try that when my wife is asleep. Lol! I have always been able to do dirty flash and it works fine. I just did the downgrade only on the main/controller and it worked flawlessly.

craigbailey18, 2024-02-21:
@gnuton I didn't try that but I will test again later tonight.

robbbaxley, 2024-02-24:
@craigbailey18 , were you able to try the erase NVRAM per @gnuton ?  I haven't been able to try it out yet.

craigbailey18, 2024-02-24:
I have been busy so didn't get a chance to do it yet.

gurple, 2024-02-25:
I'm also having odd issues with the XT8s (mostly in WiFi backhaul). All devices are now running 3004.388.6-gnuton1. It seems that now the 'hidden network' that the mesh sets up to manage itself is disabled. They seem to insist on using the weak user-facing 2.4GHz network for their backhaul channel.

[image]

[image]

gurple, 2024-02-26:
Just to be clear, before reporting this I did perform the full WPS-button-pushing reset. However, I didn't think to check the various radio settings of each XT8 after re-adding them to the network. My bad.

I'm back on the official firmware for these XT8s until I have more time to poke around. Please let me know if there are any specific things I should directly check at that time.

robbbaxley, 2024-02-26:
I tested the new @gnuton pre-release, 3004.388.6_0-gnuton1, this morning on my XT8's and the remote/slave nodes updated OK without any obvious issues (wired backhaul). However, the primary/master node still exhibits the same issue where, when the 3004.388.6 firmware is loaded, all remote/slave nodes become inaccessible (invisible) to the primary/master XT8 node.

First, I did a dirty upgrade, then for my second attempt, I performed a 'clean' NVRAM upgrade, and there was no difference for me. Also, for the first time, downgrading failed until I performed a NVRAM erase. I was then able to upload the old configuration and perform a downgrade to the 3004.388.5_1 Gnuton firmware without any further issues.

So right now, my remote/slave XT8 nodes are running the pre-release, and my primary node is on the previous stable @gnuton 3004.388.5_0-gnuton1.

There's definitely something going on with the XT8's and this firmware version.

craigbailey18, 2024-02-26:
Yes I finally had a chance to try again this morning and I have the same issue on the primary node. I have the pre-release on the nodes and the official on the primary node.

robbbaxley, 2024-02-27:
@gnuton ....do you have any idea what could be going on here? It appears that the XT8's have an issue with this version of the firmware on the primary/master node.

gnuton, 2024-02-27:
No idea yet, I will update u as soon as I know the problem

Dodgydrains, 2024-02-27:
@robbbaxley I have 4 x XT8's and dirty flashed 3004.388.6_0-gnuton1 on all the units over official FW 3.0.0.4.388_24609.  Haven't experienced any issues whilst flashing.  I always flash the furthest away node first and work my way up to the the primary router.

robbbaxley, 2024-02-27:
@Dodgydrains , yes that's my exact procedure as well. One thing I do notice is that all of us are in "AP" mode and have "Wired Backhaul". Is that your setup too? If it is, I will try what you did and dirty flash over the Asus original firmware. I did try a dirty and clean NVRAM flash...both failed for me but I was coming from the previous @gnuton version.

Dodgydrains, 2024-02-28:
@robbbaxley My primary router is connected to my ISP provided modem.  All my XT8's are wireless backhaul.  Could that be the difference as you are wired backhaul?

robbbaxley, 2024-02-28:
@Dodgydrains , it's certainly possible and I hope it points @gnuton in the directly of whatever has changed between this version and the previous version to cause this issue.

gnuton, 2024-02-28:
@robbbaxley flashing stock and back makes config just more dirty and unpredictable.
Have you tried to reset to factory the master and one node and try to discover them again? What has happened? This is what has to be tested at first.

FlashLight34, 2024-02-28:
Me i have wifi backhaul(2x xt8) and the problem is the second xt8 (node) not connect and flashing blue

robbbaxley, 2024-02-29:
@FlashLight34 , it does _appear_ that this issue/bug is related somehow. I am going to attempt @gnuton's latest request to reset all of the devices and set them up as if they were all brand new but using the latest firmware.

gnuton, 2024-02-29:
Ran a test yesterday and my xt8 was able to connect to the node via 2,4 GHz, but not via 5ghz. Maybe the issue is there. No idea why this happens

robbbaxley, 2024-03-01:
@gnuton @FlashLight34 @craigbailey18 @Dodgydrains @gurple 

Ok, so I did the last thing @gnuton recommended, which is defaulting the firmware of all devices and resetting up the system from scratch.  What I did tonight was:

1. Default all 3 XT8s (1 master and 2 slave nodes) and both XT9s (both slave nodes).
2. All XT8 slave nodes were already on the latest version of gnuton's 388.6 firmware, Master node was on 388.5. XT9's are running default Asus firmware 3.0.0.4.388.23285
3. I upgraded the Master node to 388.6
4. I then re-setup the Master and 1 each of the XT8 and XT9 slave nodes in Router mode.
5. Everything works as expected. No issue noted so far.

In the coming days I will move the system back to AP mode from Router mode to test to see if that was the problem...unless someone wants to try that first?  :D

craigbailey18, 2024-03-03:
@gnuton @FlashLight34 @Dodgydrains @gurple 
So I reset everything and set it up from scratch with the official 388_6 firmware in AP mode and it worked. It was a bit time-consuming since I had many custom settings but it finally worked. I am wondering if this has to be done with future firmware or if was it something different in 388_5 to 388_6 that didn't jive with one another.

Dodgydrains, 2024-03-03:
@craigbailey18 good to hear it worked for you.  Always best to do factory reset if you encounter issues to rule this out.

craigbailey18, 2024-03-03:
Yup

craigbailey18, 2024-03-03:
This success was short-lived as I had an issue with one of the nodes and tried to set everything up from scratch and had no luck. I reset the units several times and I got them to connect wireless and then switched one node to wired backhaul all in router mode. As soon as I switched back to AP mode the main node would not detect the secondary nodes. I guess I will be forced to revert to 388_5 unfortunately.

robbbaxley, 2024-03-03:
@craigbailey18 I just realized that I had also setup my XT8's in router mode and didn't test them in AP mode. Since you are experiencing the same issue with the XT8's in AP mode even after you reset everything to default and confirmed router mode to work, This makes me believe the issue lies with the 388.6 code. Hopefully, @gnuton can track the issue/bug down and squash it!  :)

craigbailey18, 2024-03-03:
@robbbaxley I definitely think it is the code as well. I am hoping @gnuton can track the bug as well.

hulleyrob, 2024-03-10:
wish id read this first, upgrade to 388.6 and spent nearly 3 hours trying to get the mesh to work before reverting back to 388.5 and it working instantly.

craigbailey18, 2024-03-10:
@hulleyrob I am sorry to hear it happened to you as well. I am hoping the new 388_7 addresses the bug.

quangkieu, 2024-03-19:
For my setup as in picture 

<details>
<summary>Details</summary>

[image]

</details>

All mesh routers have direct LAN to a main switch then to main router but they still show using WIFI or LAN to second router.
But when I did speedtest, it was full 700-800mbs. So, I think they do have LAN backhaul but the graphic just messes up and they think it is in WIFI mode or do not have direct LAN to main router.

ajayramaswamy, 2024-03-20:
Are you sure you are using gnuton firmware on a ZenWiFi XT8 / RT-AX95Q? The picture shows a ET8 and a non gnuton firmware

hulleyrob, 2024-04-02:
Thanks for the warning I hadn't got round to trying it out yet so I can save the hassle and wait for a new release.

craigbailey18, 2024-04-02:
You're welcome @hulleyrob

CHF-UK, 2024-04-05:
> Is this affecting et8 too?

I have 4x ET8's in router mode which connected wirelessly. Just dirty flashed 3004.388.6_2-gnuton1 and no obvious issues.

adampk17, 2024-04-10:
I have 3 ET8's and I seem to be experiencing this issue on 3004.388.6_2-gnuton1.

adampk17, 2024-04-11:
> I have 3 ET8's and I seem to be experiencing this issue on 3004.388.6_2-gnuton1.

I can confirm downgrading to 3004.388.5_0-gnuton1 on my ET8's allowed me to set up AIMesh.

CPngN, 2024-04-11:
Does it work ok with Ethernet link?
I gave up on WiFi between the 2 ET8s and bought some TP-Link power line
gigabit adapters. Has done wonders to keep upstairs stable.

adampk17, 2024-04-11:
> Does it work ok with Ethernet link? I gave up on WiFi between the 2 ET8s and bought some TP-Link power line gigabit adapters. Has done wonders to keep upstairs stable.
> […](#)

I am using a wired backhaul currently and it's working as expected.

CPngN, 2024-04-12:
Awesome, thank you.  Does 388.6_2 feel any "better" or more stable than
388.4?

Dodgydrains, 2024-04-12:
For my use case 388.4 was more stable than 388.6.  I had issues all day today with calls made over Microsoft Teams.  Reverted back to stock official Asus FW to resolve this problem.  I will wait till the next release.  Looks like 388.6 is not working well on my XT8 setup.

CPngN, 2024-04-25:
Bummer, but thanks for the update. I've had to reboot every few days
lately. It feels like there's a memory leak and if everyone's doing a
certain type of streaming And/or game playing it eventually the chokes
itself. But I also have these TP-Link power line adapters which also need
to be re-plugged when they've seen heavy traffic. So yeah. Seems to be a
common issue...? Every time I look to another brand to try to overcome this
need to reboot weekly, The comments from those customers make me happy that
I'm where I'm at. Except I'm not happy with these ET8s either. LOL I mean
problem is probably choosing the wrong hardware for 50 to 60 devices
connected simultaneously... 😞

adampk17, 2024-05-09:
Has anyone tried 3004.388.7_1 to see if this issue is resolved?

craigbailey18, 2024-05-10:
Not yet resolved. I am not sure it will be anytime soon. You may just have to leave the old firmware on the master node and user the new firmware on the points for now.

adampk17, 2024-05-11:
> Not yet resolved. I am not sure it will be anytime soon. You may just have to leave the old firmware on the master node and user the new firmware on the points for now.

Why do you think it won't be fixed any time soon?

craigbailey18, 2024-05-12:
Well the bug has been present for close to 3 months that's all I was saying. I appreciate all the hard work that @gnuton does but this seems like a more complex.

gnuton, 2024-05-13:
AIMesh is a closed component and it come says prebuilt binary from Asus.  
I was not able to spot issues on my side. 
As for now my understanding is this will be fixed with the next GPL. The other option is to downgrade one part of the firmware to the working version of the prebuilt. But this not always is possible and I have not actually checked this order yet. Another option so to use the xt8 prebuild, but I am not even sure if that's a good idea, since mismatching prebuilts can add instability on other side.

Tldr I am waiting for a fix from ASUS.

craigbailey18, 2024-05-13:
I understand @gnuton. That's why I was recommending to just use the older firmware for now.

gnuton, 2024-05-20:
INVESTIGATION:

The reason why there are no mesh nodes dispayed comes from http://192.168.50.12/require/modules/amesh.js
get_cfg_clientlist.length is undefined

an hook in web.c is defined for this, this somehow returns null.

hulleyrob, 2024-06-10:
Is this expected to be fixed in the latest release?

gnuton, 2024-06-11:
No this should be still broken since it s based on the same Asus GPL

ajayramaswamy, 2024-06-18:
@Gnuton, in the release notes for the ASUS firmware version  3.0.0.4.388_24621 dated 2024/03/19 there is this paragraph 

> *Please be advised that due to a security upgrade in AiMesh, we strongly recommend against downgrading to previous firmware versions, as this may lead to connection issues. Should you encounter any difficulties, resetting the AiMesh router to its default settings and re-establishing the mesh connection can resolve the problem.

do you know if the current gpl source and the binary from 3.0.0.4.388_24621 match up?

KristianLH, 2024-06-28:
Can anyone please advise if the latest stable release from june 10 [3004.388.7_1-gnuton1] is safe to install on AX95Q (XT8) - or if this issue still persists?

Turttle, 2024-06-30:
Additional update,  if the Main node is downgrade to 3004.388.5_1-gnuton1 and the mesh node are kept at 3004.388.7_1-gnuton1, the AI mesh works fine and is able to connect in AP mode and using ethernet backhaul configuration.

hulleyrob, 2024-08-06:
> No this should be still broken since it s based on the same Asus GPL

"UPDATED: backported all upstream changes up to 3004.388.8_2" does this include a newer GPL? Sorry to ask but its not obvious to me.

Hilipatipippan, 2024-08-09:
> > No this should be still broken since it s based on the same Asus GPL
> 
> "UPDATED: backported all upstream changes up to 3004.388.8_2" does this include a newer GPL? Sorry to ask but its not obvious to me.

Does not work for me. Two XT8 with WiFi backhauling, in AP mode.

areyouben, 2024-08-18:
I experienced the same issue after dirty flashing from 3004.388.5_0 to 3004.388.8_2. I solved it by doing a factory reset on both the parent and the node. Moved the node close to the parent and they were able to talk to each other again (over Wifi). I moved the node back to its own spot, changed the mode back to ethernet backhaul and it kept working. So although normally a dirty flash works while it is not recommended, you could try a factory reset to solve this issue. Takes a bit more time, though.

Hilipatipippan, 2024-08-18:
> I experienced the same issue after dirty flashing from 3004.388.5_0 to 3004.388.8_2. I solved it by doing a factory reset on both the parent and the node. Moved the node close to the parent and they [...]

Hello,
Did not work for me....

adampk17, 2024-10-01:
@gnuton Would it be possible to get a summary of this bug that a lay person can understand, especially in light of the length of time this issue has been unresolved. 

Thank you

[27 more replies]