gnuton/asuswrt-merlin.ng #545: RT-AX92U - Error with LED on AiMesh node

Sharpik66, 2024-02-29:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
3004.388.4_0-gnuton1; 3004.388.5_0-gnuton1; 3004.388.6_0-gnuton1

**Describe the bug**
On the main node, the LEDs turn off after a reboot, but on the AiMesh network node, the LEDs continue to light after a reboot

gnuton, 2024-03-01:
what's the expected and actual behaviour?

mirec2283, 2024-03-04:
The same here.
I use a TUF-AX5400 as the main AP and an RT-AX5400 as a node.
Both are running on same and latest firmware (3004.388.6_0-gnuton1).
After a reboot (for example, a scheduled reboot), the LEDs on the node turn on even though they had been disabled before the reboot.
The LEDs on the main AP remain off. The issue is that the node does not retain the LED disabling setting.

LED disabling is configured in:
AiMesh -> "selected node" -> Management -> LED (switch off)
Administration -> System -> Disable LEDs -> Yes

TLDR: 
It seems that only the node fails to remember the stored setting to keep the LEDs turned off,

ExplorerNL, 2024-07-10:
Same issue here, latest stable version for AX92U, LEDs on node does not remember the settings when turned off and on. LEDS turn ON after boot even when settings shows OFF

ExtremeFiretop, 2024-07-10:
@gnuton 

Same issue here, latested alpha for 3004.388.8 both me and @ExplorerNL tested and reported in the discord "support" channel.
I noticed this https://github.com/gnuton/asuswrt-merlin.ng/issues/506 however the test builds are no longer available.