gnuton/asuswrt-merlin.ng #546: Cannot flash DSL-AX82U with GNUton firmware

Eclipse-11, 2024-03-02:
**Router Model Affected**
DSL-AX82U

**Firmware Version Affected**
DSL-AX82U_3004_388.6_0-gnuton1_puresqubi.w
Also tried with previous stable version:
DSL-AX82U_3004_388.5_0-gnuton1_puresqubi.w

Asus firmware:
DSL-AX82U_3.0.0.4_388_23756-gc732440_cferom_puresqubi.w

**Is this bug present in upstream Merlin releases too?**
N/A. Only gnuton firmware available for this model.

**Describe the bug**
Cannot flash DSL-AX82U with GNUton firmware

**To Reproduce**
Steps to reproduce the behavior:

1. Connect from Windows PC to DSL-AX82U LAN port via wired connection
2. Reset DSL-AX82U to factory default.
3. Uploaded latest GNUton firmware (have tried via both browser/GUI as well as the Asus firmware restoration tool)
5. Upgrade progresses to 100% and no errors occur, but after reboot the original Asus firmware is still there.
6. I also tried the previous stable version of GNUton. No luck.

**Expected behavior**
Router should boot with gnuton firmware.

Also posted for help here but have not been able to find a fix:
https://www.snbforums.com/threads/cannot-flash-dsl-ax82u-with-gnuton-firmware.89092/

gnuton, 2024-03-02:
Please paste the logs. Not ak issue with the firmware itself since btw. Logout and login again, reset to factory and flash via cable.

Eclipse-11, 2024-03-02:
[syslog (1).txt](https://github.com/gnuton/asuswrt-merlin.ng/files/14470996/syslog.1.txt)
[syslog (2).txt](https://github.com/gnuton/asuswrt-merlin.ng/files/14470997/syslog.2.txt)

Hi. Logs attached. I reset to factory default and flashed via a cable.

gnuton, 2024-03-03:
are these logs used when you tried to flash?

Eclipse-11, 2024-03-06:
> can you flash an older asus stock firmware?
Guess what... it worked!

I flashed to the second most recent Asus firmware (DSL-AX82U_3.0.0.4_388_22048-gf313579_cferom_puresqubi.w)

Then I flashed to latest GNUton firmware (DSL-AX82U_3004_388.6_0-gnuton1_puresqubi.w) and it worked immediately! I restored my saved config and I'm up and running with GNUton. Great stuff.

Thanks for the suggestion. I'm a bit PO'd at myself because I did think of this but forgot about it before actually giving it a try. Much appreciated.

But that also means there is some sort of compatibility issue between the latest Asus and GNUton firmware. So I guess we keep this issue open for now.

gnuton, 2024-03-06:
there is nothing I can do for this. there are no compatibility issues. there was something wrong on your unit. I am not sure what's t the problem you can attach a serial TTL and get the logs of what was failing, if you notice this again you may have some hardware issues.

closed 2024-03-06 by gnuton