gnuton/asuswrt-merlin.ng #548: Dnsmasq version is old (CVE 2023-50868 and CVE 2023-50387)

Jacky10X, 2024-03-05:
Model: TUF-AX3000 
Firmware version: TUF-AX3000_3004_388.6_0-gnuton1

Hello, Sorry to bother you.
Gnuton build is currently using Dnsmasq 2.89 version if i check it right. It seems this version does not resolve the CVE 2023-50868 and CVE 2023-50387 issues.

Asuswrt merlin update the dnsmasq to 2.90 to resolves CVE 2023-50868 and CVE 2023-50387
[Merlin 3004.388.6_2, they also fix the XT12's LACP missing problem on that version]

Do you have any plan to update the dnsmasq in the future?

gnuton, 2024-03-05:
Usually these updates are covered from upstream. @RMerl

gnuton, 2024-03-08:
Yes I will release soon

Jacky10X, 2024-03-08:
@gnuton hello, that is good new but dnsmasq has 2.91 version which fix spurious "resource limit exceeded messages" issue.

RMerl, 2024-03-08:
2.90 is still the latest version available as of today.  Simon hasn't released a new update yet following the post-2.90 fixes.

That fix was already [backported](https://github.com/RMerl/asuswrt-merlin.ng/commit/e9fea56583aebe465a4a57249bdfd3cdd9f2f5d1) to my repo (as well as other post-2.90 fixes available upstream), so that fix will also be included whenever @gnuton merges my code in.

Jacky10X, 2024-03-08:
@RMerl thank you for answering and asuswrt-merlin. So fixes will be included in next update. That is also great news.

gnuton, 2024-03-08:
@Jacky10X the code has been merged today and I have released the _2 alpha.

gnuton, 2024-03-17:
Since the alpha are out.. I will close this.. 
this will naturally be part of the next stable when out. so no actions nor anything to track.. just wait update the the latest pre and stable releases.

closed 2024-03-17 by gnuton