gnuton/asuswrt-merlin.ng #550: 🌟 Support for RT-AX58U V2 🌟

gnuton, 2024-03-09:
*This thread is about supporting RT-AX58U V2.
If you see this issue here is because I have decided to support this model.

So, feel free to ask questions here about this topic.
I will keep the status updated.

Status:
- [x] received GPL
- [x] received development unit
- [ ] Merged GPL to feature-branch
- [ ] Initial testing & Bugfixing phase
- [ ] Public pre-alphas
- [ ] Alpha released
- [ ] collecting testing reports  & Bugfixing phase
- [ ] Beta released
- [ ] collecting testing reports  & Bugfixing phase
- [ ] Stable released

gnuton, 2024-03-12:
looking for pre-alpha testers.
Requirement to be an alpha tester.. be confortable with soft-bricks and restoring using asus restoration tool

gnuton, 2024-03-15:
Pre-alpha available here https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8293622678/artifacts/1329003666

Stellar1616, 2024-03-16:
I just got a RT-AX3000, and it's a V2. :( I was hoping to get it working with Merlin, but I guess it's not ready yet. That being said, I'd be willing to be an alpha tester.

Souler, 2024-03-16:
I flashed the image before you removed the link and bricked my router 😅.

I was able to flash back a stock firmware, so I'm ready for testing future builds!

gnuton, 2024-03-16:
@Souler those are soft bricks something you can quickly recover.. :D

gnuton, 2024-03-22:
Hi @Stellar1616 the alpha is not  out yet. There was a pre-alpha but it was not booting.
I do not have physical access to the device now so I cannot debug what's wrong.
 if anyone is able to get the logs from the TTL attaching some jumpers on the MB that would help.

gnuton, 2024-03-24:
New pre-alpha image available
Version: pre-alpha-2
Risk: HIGH - This image has not been tested so worst case scenario your router won't start anymore and ASUS restoration tool is required. So before flashing this make sure you have a copy of the latest stock firmware.

Link to image:
https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8403778904/artifacts/1352759145

neil1962, 2024-03-24:
Link is broken

Neil
Sent with Spark

gnuton, 2024-03-24:
@neil1962 -_- the link works for me. Are you logged in github when downloading?

gnuton, 2024-03-26:
The pre-alpha seems to boot. So it should work  now.
[image]

closed 2024-04-05 by gnuton

Stellar1616, 2024-04-20:
It says at the top that the stable has been released, but I'm not showing the V2 listed as a supported model on https://www.asuswrt-merlin.net/about and on the download page on source forge, I'm only showing an entry for RT-AX58U, doesn't specify a hardware version https://sourceforge.net/projects/asuswrt-merlin/files/RT-AX58U/
But the latest stable is from 2-27-2024, so that can't be it. Do you have a link to download the firmware?

Grumpy-Geoff, 2024-04-21:
> Do you have a link to download the firmware?

It's a GNUton build, available from here -
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.6_2-gnuton1

aidancov1, 2024-05-31:
I was pleased to find support for the AX58U V2 model and all seemed well but I cannot get the VPN tunnel to block devices when it goes down. I have checked the VPN director and all seems okay. I can connect to my VPN but when I disconnect it,  my devices revert to WAN which is why I wanted to use the kill switch. Any idea why this is occurring?

Thanks

Aidan

heythem500, 2024-08-19:
I tried 
[RT-AX58U_V2_3004_388.8_2-gnuton1_nand_squashfs.pkgtb ] for some hours but I noticed when I log in to wifi my device isn't detected, so please check this point
"
Date: August 04, 2024
Release: 3004.388.8_2-gnuton1
"

thanks for the big effort man
good luck

MNLierman, 2024-08-24:
Hi, I'm really happy that this router is supported, however, it has the exact same damn bug as the original Asus code, which is that you cannot disable Roaming Assistant. The setting in the GUI does not work, as soon as a wifi client reaches -68 -69 -70 RSSI, it kicks a cold hard swift kick in the pants. Being as this is the only router and I dont' have multuiple of these, as assumed by Asus and their AiMesh (reason this feature exists) the client device keeps attempting to reconnect and reports to the router it has -50 RSSI as reported by the wifi bulbs in question which have no antenna. Router reassess the RSSI within several seconds and gives another quick hard kick. Over and over and over.

IF I enable Roaming Assistant (Professional tab of wifi) for 2.4Ghz and give it a value of -90 (max) or anything else that I wish, save the settings, and then reboot (or unplug), the result is the same. The roaming assistant MUST be using a hard coded default variable number of -70 somewhere. I know next to nothing about this programming language or I'd trying to help find it.

Again, this is a bug in the original Asus code, and it exists also in this firmware. I was hoping you all caught it and fixed it. This is severely impacting my network and it explains everything that is wrong with my smart home devices. My cameras, doorbell cam outside, outside cameras, all have a naturally weaker signal that is unfortuntely in the -70s, and that's normal. I don't have a ethernet connected home, or I would buy another routher and do as Asus wants. I really don't like it when companies force things on us that we don't won't. In the rush for "AiMesh" and Ai everything, Asus was in too big of a hurry and didn't link the actually systemd script variables that runs the wifi roaming assistant to the user GUI, or whatever it is (assuming this is Linux).

MNLierman, 2024-08-24:
Another Bug:
"Disassociated due to inactivity"

Checking the lights in question, YUP ANOTHER GD BUG by Asus. Trying to control that light which was connected less than an hour ago (remember I just rebooted the wifi and also power cycled the router so many times in the last 2 hours), Google says "Sorry it looks like Dining Light 2 isn't available right now" God fuck.