gnuton/asuswrt-merlin.ng #551: RT-AX82u cannot flash original firmware while on aiMesh

maddler, 2024-03-09:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
3004.388.6_0-gnuton1
3004.388.6_2-gnuton0_alpha1

**Is this bug present in upstream Merlin releases too?**
RT-AX82U unsupported on Merlin

**Describe the bug**
I've got my RT-AX82U configured as an AiMesh node and was trying to debug an issue with some of my client.
I tried to flash the latest original firmware, uploaded and went thru the updated process but when the device comes back the old version is still in use. I was able to correctly flash Gnuton0 versions so it, somehow, seems to be only an issue while trying original.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the update page
2. Click on update and authenticate in the new window
3. Upload the original firmware package
4. Old firmware version still being reported

**Expected behavior**
New firmware running

Thanks.

gnuton, 2024-03-09:
not sure what has happened but the log should have something. 
Not a bug of the firmware if you can flash something.

maddler, 2024-03-09:
I'll try again and double check if in the master's logs I've got anything.
That's just a bit weird that flashing another gnuton release works fine but fails with official.
Thanks anyway for your work!

gnuton, 2024-03-09:
I would be surprised if the master would hav any log about this. 

The flasher flashes raw bytes. 
Just try to flash again the OFW but check its md5.

maddler, 2024-03-09:
Will do. By the way, this looks to be happening only when the device is configured as AiMesh node, I've done that countless times and never had an issue before.
I'll give ago later and will update.
Grazie!

gnuton, 2024-03-09:
worst case scenario use the asus restoration tool to flash back to original fw

maddler, 2024-03-09:
So, just tried again, nothing in the logs (other than the fact, for some reason, the NTP settings are not being applied on the node, but that's a whole different story).
I'll have a go tomorrow at resetting and flashing before rejoining to AiMesh and try the other way round (from official to Gnuton).
Thanks again!

neil1962, 2024-03-10:
Pretty sure you have to flash Asus firmware version 3.0.0.4.386.50289  first then install latest version

maddler, 2024-03-10:
Ah, good shout! Just looked at the changelog for that, that seems to be the case.
WIll give a go in the morning!
Cheers!

closed 2024-03-10 by maddler