gnuton/asuswrt-merlin.ng #555: ddns is inactive

ikunfans, 2024-03-16:
**Router Model Affected**
Models: (eg: tuf ax3000)

**Firmware Version Affected**
388.6

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
I restored the settings,and the ddns status will be inactive in a few days. But the log show ddns status continues to be updated and can be used.
[image]

gnuton, 2024-03-16:
Restoring settings expecially from stock firmware is not guaranteed

ikunfans, 2024-03-18:
> Restoring settings expecially from stock firmware is not guaranteed

sorry i mean factory default. And now it not update if network changed.  I need  close and
 Open again.

huytn1, 2024-08-28:
Hello,

I got the same issue in the latest version 3004.388.8_2-gnuton1_tuf running on TUF AX3000. I had disable and enable DDNS again to make it active. 

Really appreciate your help with this issue.

andrewk123, 2024-12-04:
I'm seeing the same on 3004.388.8_2-gnuton1 on an XT8.  The only way I've been able to resolve it so far is to change the domain name to one I know is already in use (so the request fails as you would expect it would) and then change it back to the original value which then activates again correctly.

ikunfans, 2024-12-07:
> I'm seeing the same on 3004.388.8_2-gnuton1 on an XT8. The only way I've been able to resolve it so far is to change the domain name to one I know is already in use (so the request fails as you [...]

The issue was previously resolved by unbinding the DDNS in the ASUS router app. Never bind the domain in the app again; only enable it in the web UI.

andrewk123, 2024-12-07:
> The issue was previously resolved by unbinding the DDNS in the ASUS router app. Never bind the domain in the app again; only enable it in the web UI.

Thanks for that, however in mu case I have not used the App to bind the DDNS just the WebUI.