gnuton/asuswrt-merlin.ng #558: Bring back (?) WiFi radar on unsupported routers

F640, 2024-03-23:
**Is your feature request related to a problem? Please describe.**
I had been able to use WiFi Radar feature in my router (RT-AX82U v1), but in the most recent firmware update this feature seems to be missing.

**Describe the solution you'd like**
Bring back WiFi radar feature if it's possible.

**Describe alternatives you've considered**
none

**Additional context**
I know this feature is unsupported in this router, but i still want it back.

gnuton, 2024-03-24:
version of the firmware?

F640, 2024-03-25:
I don't remember it, but last time i used this feature around 388.2 version, probably. Just wanted to revisit this feature 2 days ago and that feature has gone.

gnuton, 2024-03-25:
I meant what's the version of the firmware you have now tested that doesn't have this feature?

F640, 2024-03-25:
Sorry about that, here it is: 3004.388.6-gnuton1

gnuton, 2024-07-15:
Wifi Radar has been removed from all builds because it was unmaintained by broadcom.

closed 2024-07-15 by gnuton