gnuton/asuswrt-merlin.ng #559: Skynet not working on TUF-AX3000 V2

gnuton, 2024-03-24:
[image]
Skynet not able to run on TUF-AX3000_V2

gnuton, 2024-03-24:
Linked to https://github.com/gnuton/asuswrt-merlin.ng/issues/123

gnuton, 2024-03-24:
Linux kernel configs 

release/src-rt-5.04axhnd.675x/kernel/linux-4.19/config_base.6a.62118G:# CONFIG_IP_SET is not set
release/src-rt-5.04axhnd.675x/kernel/linux-4.19/config_base.6a.6750:# CONFIG_IP_SET is not set <--- THIS
release/src-rt-5.04axhnd.675x/kernel/linux-4.19/config_base.6a.4912:# CONFIG_IP_SET is not set
release/src-rt-5.04axhnd.675x/kernel/linux-4.19/config_base.6a.6756:# CONFIG_IP_SET is not set
release/src-rt-5.04axhnd.675x/kernel/linux-4.19/config_base.6a.47622:# CONFIG_IP_SET is not set
release/src-rt-5.04axhnd.675x/kernel/linux-4.19/config_base.6a.6855:# CONFIG_IP_SET is not set
release/src-rt-5.04axhnd.675x/kernel/linux-4.19/config_base.6a.6756_nowl:# CONFIG_IP_SET is not set

gnuton, 2024-03-24:
Test images
https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8412377915
Note the link works only if logged in github

l0r3n75, 2024-03-29:
Same bug still occuring

gnuton, 2024-03-29:
new test images here
https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/8478486819

l0r3n75, 2024-03-29:
Experiencing some weird errors still can't start skynet
[image]

gnuton, 2024-03-30:
@l0r3n75 cannot see the image. please come with logs

gnuton, 2024-04-14:
@l0r3n75 any update?

l0r3n75, 2024-04-15:
I couldn't get it working

blooddrunk99, 2024-05-07:
I am having a similar issue on rt- ax58v2. I would be happy to provide log, I am unaware how to get log.

chaosmaou, 2024-05-20:
Same issue on the RT-AX3000 V2, which uses the RT-AX58U_V2 firmware. Happens in the latest stable 3004_388.6_2-gnuton1. Fairly sure it's the exact same issue as this, hence I didn't post a duplicate.

Adamm00, 2024-06-03:
This is also the case for the GT-BE98 (src-rt-5.04axhnd.675x)

novusB, 2024-06-05:
Same issue confirmed. I don't have the TUF but I do have the V2 version.

_____________________________________________

 Skynet install aborted,
 IPSet extension not supported, please
 update router to latest Firmware version.
_____________________________________________

 Enter amtm option  i
 amtm 4.8 FW               by thelonelycoder

 ASUS RT-AX3000 HW: armv7l Kernel: 4.19.183
 FW: 3004.388.6_2-gnu..

gnuton, 2024-06-09:
Please check these test images and let me know if it owrks now https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/9434553478

gnuton, 2024-06-09:
Fixed works
[image]

gnuton, 2024-06-09:
The fix will be part of the next release or please check the images above

closed 2024-06-09 by gnuton