gnuton/asuswrt-merlin.ng #561: Connection drops from one client  -  TUF AX5400

AMDFenics, 2024-03-26:
**Router Model Affected**
Models: TUF AX5400

**Firmware Version Affected**
3004.388.6_2-gnuton0_alpha1_tuf

**Is this bug present in upstream Merlin releases too?**
YES can't remember that this hasn't been there.

**Describe the bug**
My Weatherstation disconnects very frequently. Every 20-40 seconds.
Also i don't know, if this also related. My phone is connected to wifi, but the symbol is flashing with a "!" inside and no data is flowing.
Wifi off/on helps to get rid of that.

Weather Station is a Bresser 4Cast Pro Wifi
Phone is a Pixel 7 Pro

**To Reproduce**
Can't activly reproduce. 
The weatherstation bug is always there.
and the phone bug happens when i switch on my FireTV Stick, TV and AV Receiver!?

If you need more information i would be glad to help. Please be gently with me, it's my first bug report.

**Expected behavior**
No connection drops.

**Screenshots**
`Mar 26 20:18:41 wlceventd: wlceventd_proc_event(722): eth5: Assoc ***Wetterstation***, status: Successful (0), rssi:-54
Mar 26 20:18:41 hostapd: eth5: STA ***Wetterstation*** IEEE 802.11: associated
Mar 26 20:18:41 hostapd: eth5: STA ***Wetterstation*** RADIUS: starting accounting session DE0B3952F41FA329
Mar 26 20:18:41 hostapd: eth5: STA ***Wetterstation*** WPA: pairwise key handshake completed (RSN)
Mar 26 20:18:41 dnsmasq-dhcp[19425]: DHCPDISCOVER(br0) ***Wetterstation*** 
Mar 26 20:18:41 dnsmasq-dhcp[19425]: DHCPOFFER(br0) 192.168.100.113 ***Wetterstation*** 
Mar 26 20:18:41 dnsmasq-dhcp[19425]: DHCPDISCOVER(br0) ***Wetterstation*** 
Mar 26 20:18:41 dnsmasq-dhcp[19425]: DHCPOFFER(br0) 192.168.100.113 ***Wetterstation*** 
Mar 26 20:18:41 dnsmasq-dhcp[19425]: DHCPREQUEST(br0) 192.168.100.113 ***Wetterstation*** 
Mar 26 20:18:41 dnsmasq-dhcp[19425]: DHCPACK(br0) 192.168.100.113 ***Wetterstation*** 
Mar 26 20:18:41 dnsmasq-dhcp[19425]: DHCPREQUEST(br0) 192.168.100.113 ***Wetterstation*** 
Mar 26 20:18:41 dnsmasq-dhcp[19425]: DHCPACK(br0) 192.168.100.113 ***Wetterstation*** 
Mar 26 20:18:59 wlceventd: wlceventd_proc_event(645): eth5: Deauth_ind ***Wetterstation***, status: 0, reason: Station requesting (re)association is not authenticated with responding station (9), rssi:0
Mar 26 20:18:59 hostapd: eth5: STA ***Wetterstation*** IEEE 802.11: disassociated
Mar 26 20:18:59 wlceventd: wlceventd_proc_event(662): eth5: Disassoc ***Wetterstation***, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0`

If you need more input, i would be glad to help solving this issue.
Thank you for your time an effort!

gnuton, 2024-03-30:
please reset to factory the firmware then come back if you still have issues.

AMDFenics, 2024-04-02:
Thank you, i reset the config and will see, if its better. :)

AMDFenics, 2024-05-29:
I just want to add some news. The problem with my phone is gone. The reason was my FireTV stick. When this connected to my Wifi, my phone got the described behavior with no connection to my network. Since the FireTV stick is connected via LAN, the problem is gone. 

Whats still happening is the connection drop from my weather station. Every ~20 sec a reconnect...
I'm going to believe, that the Station itself is buggy....