gnuton/asuswrt-merlin.ng #564: DSL-AX82U 0.0 dB problem.

efsanesro, 2024-03-30:
**Modem = DSL-AX82U
Current Version : 3004.388.6_2-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
Yes

I get less upload speed because of the fact that it says 0.0 DB. could you please help me.**

**2. there is a -6 dB setting in the picture, I wonder if there is a chance to add this setting up to -20.**

[image]
[image]

Katana1074, 2024-03-31:
Hi there bug confirmed....

[image]

fards, 2024-03-31:
it is showing the same in 386_1 but there's no loss of speed here. 
So I'm not sure it's much of the reason? 

[image]

fards, 2024-03-31:
Just realised - I'd had SNR at +4 while experimenting. 

here's snr margin disabled - it does show 0.db but I don't think it's affecting anything. 

[image]

gnuton, 2024-03-31:
the attenuation value is fetched from 
```void get_xdsl_info(XDSL_INFO *info);```
the  funciton  is defined in dsl-broadcom.o. To fix this I may need to revert back this to an older versoin, but this doesn't sound good. 

The DSL-AX82U comes with two DSL firwares. Setting SNR is not supported anymore in the latest firmware, since it's considered legacy non standard by vendors (broadcom). If you set SNR on then you will use old dSL firmwares. 
The modem doesn't run merlin but it rather runs one of these two firmwres. As for now this attenuation missing value seems to be an issue from the dsl-broadcom.o
For this reason I would not be worried about this. The actual attenuation values are  written

gnuton, 2024-03-31:
I will leave this issue as problem.
ATM I do not have fix or it.