gnuton/asuswrt-merlin.ng #567: Status of latest 386.x release for DSL-AC68U ?

mackonsti, 2024-04-02:
Hello @gnuton hope you are well !

The last available 386.x build remains in Beta from last September 2023:
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.12_0-gnuton0_beta1

Of course the Stable one is there, dating over one year (January 2023):
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/388.1_0-gnuton1

I am sure that you and the guys have much more interesting things to work on, especially after ASUS leaving out DSL-AC68U from the new 388 branch/releases, so this message is really a status request on the fate of the 386.12 release and if you think it will see the light of day?

Thank you once more for bringing life to the DSL-AC68U longer than Asus did 🙏 Warm regards.

gnuton, 2024-04-03:
Stable image for testing
https://app.circleci.com/pipelines/github/gnuton/asuswrt-merlin.ng/2002/workflows/b256faef-b656-43d2-b17f-22d89f093bc1/jobs/6360/artifacts

gnuton, 2024-04-03:
Versoin 13
Status: Beta
- please test
New betahttps://app.circleci.com/pipelines/github/gnuton/asuswrt-merlin.ng/2003/workflows/b4898823-029e-45de-b8b6-23bae89cfc8f/jobs/6362/artifacts

mackonsti, 2024-04-06:
Thank you @gnuton you are the best. Sorry for my original message but it look curious, was hoping you hadn't abandoned our old beloved DSK-AC68U 🤣 Thank you again, grazie y ciao amico!

closed 2024-04-06 by gnuton

mackonsti, 2024-04-15:
Hi @gnuton how are you? I have now access to my DSL-AC68U and will update to 386.12_0 version.

Please know that the Admin interface does not show any updates available, last time it was showing the (wrong) update you had to fix something on your end, is this still the case, mate? Thanks in advance.

[image]

gnuton, 2024-04-15:
Thanks for reporting. I will check that

sopla4ever, 2024-06-02:
Hellow @gnuton, I saw that the code for 386.13_2 shows as stable as of April 26. Could you please build and release this version?  
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.13_2

mackonsti, 2024-11-30:
Hi @gnuton how are you?
Apologies to piggyback on this ticket -- Any possibility of releasing latest 386 build for us old DSL users? 😄 
Would [Unstable: 386.14_0-gnuton0_beta1](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.14_0-gnuton0_beta1) be ready as stable, you think?
Grazie!