gnuton/asuswrt-merlin.ng #573: TUF AX5400 Redirect webui

doggi87, 2024-04-07:
**Router Model Affected**
Models: TUF AX5400

**Firmware Version Affected**
3004.388.6_0-gnuton1 and 3004.388.6_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
Redirect webui access to www.asusrouter.com under Administration -> System is broken again. 
Redirect worked on version: 3004.388.5_0-gnuton1

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Administration 
2. Click on System 
3. Scroll down to Redirect webui access to www.asusrouter.com
4. Doesn't work if you change to yes or no

gnuton, 2024-04-25:
Hi @doggi87
Can you please link the bug where this was fixed?

doggi87, 2024-04-25:
Hi,

It was fixed in

https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.5_0-gnuton1

Regards.

closed 2024-05-05 by gnuton

doggi87, 2024-05-05:
Awesome, thanks for the update!