gnuton/asuswrt-merlin.ng #575: TUF AX5400 - Logout button not aligned

doggi87, 2024-04-07:
Router Model Affected
Models: TUF AX5400

Firmware Version Affected
3004.388.6_0-gnuton1 and 3004.388.6_2-gnuton1

Is this bug present in upstream Merlin releases too?
Don't know

**Describe the bug**
The button for Logout is not centered and if you see the picture you can click on the red marked dot and not on the text. Text should be centered.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section
[image]

gnuton, 2024-04-08:
do you see the same issue if you browse it in incognito mode?

gnuton, 2024-04-25:
Fix 1
file:///home/aaloisio/Pictures/Screenshots/Screenshot%20from%202024-04-25%2023-55-29.png

gnuton, 2024-04-25:
Fix 2
[image]

gnuton, 2024-04-25:
commit 0f811d381e7c82a756c74522a2a01c54b6575943

gnuton, 2024-06-01:
Fix b27fcb6a3026293709bba98807a8dda1b3d2638e

gnuton, 2024-06-02:
merged to master

closed 2024-06-02 by gnuton