gnuton/asuswrt-merlin.ng #577: RT-AX82U V2 not correcrtly showing installed firmware version after update

henr1quesj, 2024-04-12:
**Router Model Affected**
RT-AX82U V2

**Firmware Version Affected**
3004.388.6_2-gnuton1
(RT-AX82U_V2_3004_388.6_2-gnuton1_nand_squashfs.pkgtb)

**Is this bug present in upstream Merlin releases too?**
NA

**Describe the bug**
After updating the firmware version is not updated.

Also applied latest firmware update to TUF-AX5400 and it shows correctly the version installed as evidenced in screenshot.

**To Reproduce**
Steps to reproduce the behavior:
1.Install latest Firmware
2. http://192.168.50.1/Advanced_FirmwareUpgrade_Content.asp
3. See version not updated to show applied firmware

**Expected behavior**
Show firmware version applied

**Screenshots**
[image]

iam-black, 2024-04-13:
It seems that I have a somehow related issue. I could not install 3004_388.6_2-gnuton1_nand_squashfs.pkgtb on a stock RT-AX82U V2. The progress is completed successfully, but I did not see any version change, no UI changes and the Merlin logo did not show up as well. 3004_388.6_0-gnuton1_nand_squashfs.pkgtb installed okay

gnuton, 2024-04-14:
`ay  5 07:05:05 kernel: klogd started: BusyBox v1.25.1 (2024-01-31 17:14:18 UTC)
May  5 07:05:05 crashlog: ^A
May  5 07:05:05 crashlog: <5>Linux version 4.19.183 (docker@3380abedea93) (gcc version 9.2.0 (Buildroot 2019.11.1)) #1 SMP PREEMPT Thu Apr 4 22:58:31 UTC 2024
May  5 07:05:05 crashlog: <6>CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
May  5 07:05:05 crashlog: <6>CPU: div instructions available: patching division code
May  5 07:05:05 crashlog: <6>CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
May  5 07:05:05 crashlog: <6>OF: fdt: Machine model: RT_AX82U_V2
May  5 07:05:05 crashlog: <6>bootconsole [earlycon0] enabled
May  5 07:05:05 crashlog: <6>Memory policy: Data cache writealloc
May  5 07:05:05 crashlog: <6>Reserved memory: created CMA memory pool at 0x1f400000, size 12 MiB
May  5 07:05:05 crashlog: <6>OF: reserved mem: initialized node dt_reserved_cma, compatible id shared-dma-pool
May  5 07:05:05 crashlog: <7>On node 0 totalpages: 131072
May  5 07:05:05 crashlog: <7>  Normal zone: 1024 pages used for memmap
May  5 07:05:05 crashlog: <7>  Normal zone: 0 pages reserved
May  5 07:05:05 crashlog: <7>  Normal zone: 131072 pages, LIFO batch:31
May  5 07:05:05 crashlog: <6>psci: probing for conduit method from DT.
May  5 07:05:05 crashlog: <6>psci: PSCIv1.1 detected in firmware.
May  5 07:05:05 crashlog: <6>psci: Using standard PSCI v0.2 function IDs
May  5 07:05:05 crashlog: <6>psci: MIGRATE_INFO_TYPE not supported.
May  5 07:05:05 kernel: Linux version 4.19.183 (docker@6910448c0b91) (gcc version 9.2.0 (Buildroot 2019.11.1)) #1 SMP PREEMPT Wed Jan 31 17:14:45 UTC 2024
May  5 07:05:05 crashlog: <6>psci: SMC Calling Convention v1.0
May  5 07:05:05 crashlog: <5>random: get_random_bytes called from start_kernel+0x9c/0x490 with crng_init=0
May  5 07:05:05 crashlog: <6>percpu: Embedded 26 pages/cpu s77004 r8192 d21300 u106496
May  5 07:05:05 crashlog: <7>pcpu-alloc: s77004 r8192 d21300 u106496 alloc=26*4096
May  5 07:05:05 crashlog: <7>pcpu-alloc: [0] 0 [0] 1 [0] 2 
May  5 07:05:05 crashlog: <6>Built 1 zonelists, mobility grouping on.  Total pages: 130048
May  5 07:05:05 crashlog: <5>Kernel command line: console=ttyAMA0 earlyprintk debug irqaffinity=0 pci=pcie_bus_safe isolcpus=2 rootwait mtdparts=brcmnand.0:1048576(loader),265289728@1048576(image),524288@266338304(crashlog) root=/dev/ubiblock0_6 ubi.mtd=image ubi.block=0,6 rootfst
May  5 07:05:05 crashlog: ype=squashfs cma=0M
May  5 07:05:05 crashlog: <6>Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
May  5 07:05:05 crashlog: <6>Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
May  5 07:05:05 crashlog: <6>Memory: 496096K/524288K available (6144K kernel code, 218K rwdata, 1284K rodata, 1024K init, 302K bss, 15904K reserved, 12288K cma-reserved, 0K highmem)
May  5 07:05:05 crashlog: <5>Virtual kernel memory layout:
May  5 07:05:05 kernel: random: get_random_bytes called from start_kernel+0x9c/0x490 with crng_init=0
May  5 07:05:05 crashlog: <5>    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
May  5 07:05:05 crashlog: <5>    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
May  5 07:05:05 crashlog: <5>    vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
May  5 07:05:05 crashlog: <5>    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
May  5 07:05:05 kernel: Kernel command line: console=ttyAMA0 earlyprintk debug irqaffinity=0 pci=pcie_bus_safe isolcpus=2 rootwait mtdparts=brcmnand.0:1048576(loader),265289728@1048576(image),524288@266338304(crashlog) root=/dev/ubiblock0_4 ubi.mtd=image ubi.block=0,4 rootfstype=squashfs cma=0M
May  5 07:05:05 crashlog: <5>    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
May  5 07:05:05 crashlog: <5>    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
May  5 07:05:05 kernel: Virtual kernel memory layout:
May  5 07:05:05 crashlog: <5>      .text : 0x(ptrval) - 0x(ptrval)   (7136 kB)
May  5 07:05:05 kernel:     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
May  5 07:05:05 crashlog: <5>      .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
May  5 07:05:05 kernel:     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
May  5 07:05:05 crashlog: <5>      .data : 0x(ptrval) - 0x(ptrval)   ( 219 kB)
May  5 07:05:05 kernel:     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
May  5 07:05:05 crashlog: <5>       .bss : 0x(ptrval) - 0x(ptrval)   ( 303 kB)
May  5 07:05:05 kernel:     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
May  5 07:05:05 crashlog: <6>SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=3, Nodes=1
May  5 07:05:05 kernel:     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
May  5 07:05:05 crashlog: <6>rcu: Preemptible hierarchical RCU implementation.
May  5 07:05:05 kernel:     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
May  5 07:05:05 kernel:       .text : 0x(ptrval) - 0x(ptrval)   (7136 kB)
May  5 07:05:05 crashlog: <6>	Tasks RCU enabled.
May  5 07:05:05 kernel:       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
May  5 07:05:05 crashlog: <6>NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
May  5 07:05:05 kernel:       .data : 0x(ptrval) - 0x(ptrval)   ( 219 kB)
May  5 07:05:05 crashlog: <6>arch_timer: cp15 timer(s) running at 50.00MHz (phys).
May  5 07:05:05 kernel:        .bss : 0x(ptrval) - 0x(ptrval)   ( 303 kB)
May  5 07:05:05 crashlog: <6>clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
May  5 07:05:05 crashlog: <6>sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
[... cut ...]
May  5 07:05:05 crashlog: <4>1f60: 00000001 671a3ac0 fffffff5 de881f88 c055c258 c055c25c 00000013 ffffffff
May  5 07:05:05 crashlog: <4>[<c02020cc>] (__irq_svc) from [<c055c25c>] (cpuidle_enter_state+0x88/0x314)
May  5 07:05:05 crashlog: <4>[<c055c25c>] (cpuidle_enter_state) from [<c024b9e8>] (do_idle+0x1b0/0x218)
May  5 07:05:05 crashlog: <4>[<c024b9e8>] (do_idle) from [<c024bd2c>] (cpu_startup_entry+0x18/0x1c)
May  5 07:05:05 crashlog: <4>[<c024bd2c>] (cpu_startup_entry) from [<00202acc>] (0x202acc)
May  5 07:05:05 kernel: CSIMON:  CSIMON[1.1.0] Initialization`

bysvindal, 2024-04-18:
I can confirm as well. AX82U V2. update shows completed but the router remained on the stock fimrware

gnuton, 2024-04-20:
Before flashing the image gets mounted and started but it crash
There are some missing simbols. 

This should be a regression which was not caught by testers -__-

```
May  5 07:05:05 kernel: brcm_otp_init entry
May  5 07:05:05 kernel: emf: Unknown symbol osl_malloc (err -2)
May  5 07:05:05 kernel: emf: Unknown symbol osl_pktdup_cpy (err -2)
May  5 07:05:05 kernel: emf: Unknown symbol osl_mfree (err -2)
May  5 07:05:05 kernel: emf: Unknown symbol linux_pktfree (err -2)
May  5 07:05:05 kernel: emf: Unknown symbol osl_pktdup (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_wmf_scbfind (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol osl_malloc (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_mfdb_ipv6_membership_add (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol osl_detach (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_rtport_del (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_igmp_snooper_register (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol osl_attach (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_mfdb_ipv6_membership_del (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_wmf_get_igsc (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_rtport_add (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_igmp_snooper_unregister (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_mfdb_membership_del (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol osl_mfree (err -2)
May  5 07:05:05 kernel: igs: Unknown symbol emfc_mfdb_membership_add (err -2)
```

gnuton, 2024-04-22:
In the main page of this project you can find the answer to this question, but in a nutshell you wanna SSH on the router and run `nvram get productid`  or just from tool info if you arelady have merin
[image]

gnuton, 2024-04-22:
The alpha and beta testing for this model completely failed.. and I had nobody checking this model.
So beside fixing I think I would need to get more commitment from the community when it comes to proper testing stuff.
Said that the latest image I can flash is the latest stable 3004.388.6_0-gnuton1.

gnuton, 2024-04-24:
fix c7db2c4dc0419a590e7f35703ce2325302a247fb
I will publish new alpha with this fix in a bit, since it affects all new models.

gnuton, 2024-04-24:
Fix released in the latest alpha
Feel free to update, it should be stable enough.
[image]

closed 2024-04-24 by gnuton