gnuton/asuswrt-merlin.ng #579: Add Zram swap to firmware

YoshiKill, 2024-04-18:
Hello, can you add Zram swap function to merlin firmware? Use it from web Gui is be better way.

gnuton, 2024-04-21:
is there any proof that adding ZRam would be beneficial for small devices like these routers?

YoshiKill, 2024-04-21:
> is there any proof that adding ZRam would be beneficial for small devices like these routers?

i don't find any statistic proof that is show how Zram help or not help, but many people on openwrt with 512mb use it and happy with it, because any way its work better than usb swap with it low speed. I theory how its work, with normal power processor like in ax3000 v2 you can up your ram from 512 to 812mb with out any problem. I use it to in my owrt router with 512mb ram and its helps a lot with performance. Its really not feature for feature.

sskars2023, 2024-05-15:
Depends on the routers in which you enable it, if you are having a lot of devices connected it will help a lot as it will compress the applications in the RAM however the trade of is due to the compressing and decompressing needed if the router itself has a poor CPU you could have the opposite happen where CPU usage increases and it negatively affects the performance, nice to have the option but would be good if the feature could be disabled as well in some cases

robbbaxley, 2024-06-13:
I second this request   :)

leonpano2006, 2025-02-22:
i agree on this request

leonpano2006, 2026-08-20:
> is there any proof that adding ZRam would be beneficial for small devices like these routers?

Coming back to this with actual numbers, since I only had an opinion last time. I've been running compressed swap on a GT-BE98 (BCM6813, 4× Cortex-A53-class, 2 GB RAM) in my own builds since late July — first near-stock, then with the backports listed at the bottom (I'd also done a similar setup on a GT-AXE16000 before this box). Structured by adoption cost, cheapest first.

## Level 0 — works today, zero kernel patches

This is the part that answers the original request directly:

- **zram is already in this kernel.** 4.19 ships `zram` + `zsmalloc` + the crypto compressors (lzo/lz4, and this SDK has `crypto/zstd.c` too). Built **as modules** they cost a few tens of KB in the image and have zero runtime footprint until someone runs `modprobe zram` — which is exactly how Debian/Fedora/Arch/OpenWrt ship it. Self-contained: a compressed RAM block device, no USB stick required, works on a bare router.
- **zswap is also already there.** `zswap.enabled=1 zswap.compressor=zstd zswap.zpool=z3fold` is a boot parameter away — no backport needed to get it running. On a stock 4.19, **prefer `z3fold` (or `zbud`) as the zpool**: they implement pool reclaim, so a full pool writes back to swap instead of stalling. `zsmalloc` is denser but on this kernel it should really be paired with the Level 1 patches below. The catch: zswap is a cache *in front of* real swap, so it only helps people who already have a swap device. That's why I'd expose zram first — it's the one-step option.

So the minimum viable version of this feature request is: **flip two kernel config symbols to `m`, done.** Users who don't want it never pay for it.

## What it buys — measured

Live figures from my unit (which also runs a UniFi controller/JVM, minidlna and netdata):

```
MemTotal:      2047228 kB    (2 GB)
Zswapped:       676024 kB    original size of the pages held
Zswap:          189176 kB    what they occupy in RAM
                             → 3.57:1 — 660 MB of pages living in 185 MB,
                                        475 MB of RAM effectively recovered
compressor: zstd   zpool: zsmalloc   max_pool_percent: 30
pool_limit_hit: 0   reject_compress_poor: 0   reject_alloc_fail: 0
same_filled_pages: 17593
```

Load average at capture: `1.00 / 1.29 / 1.34` on 4 cores. PSI memory pressure: `some avg300=2.81 / full avg300=1.79` — under 3% of time stalled on memory, on a box that would otherwise be paging to a USB stick.

Three honesty caveats:

- **My zsmalloc is not stock** (see Level 2) — zram uses the same allocator, so on an unpatched tree both zram and zswap would pack somewhat worse than 3.57:1. Read it as "this workload, patched allocator", not a universal number.
- I measured **zswap, not zram** — but they sit on the same machinery (zram is hard-wired to zsmalloc in its Kconfig and calls `zs_malloc()` directly; my zswap uses zsmalloc via zpool). Same allocator, same crypto compressors, same page-sized inputs — same ballpark, and I'd rather say that plainly than imply I tested both.
- `reject_compress_poor: 0` means nothing was too incompressible to store. zram has no such escape hatch (a page past `huge_class_size` is stored whole, flagged `ZRAM_HUGE`) — irrelevant for this workload, matters on boxes full of already-compressed data.

**On the CPU-cost concern** (@sskars2023's fair point): it doesn't show up on this class of hardware. zstd compresses at hundreds of MB/s per core on these 1.5–2 GHz quad ARMv8 parts, and the real comparison is "compress a 4K page" vs "page fault to USB/flash" — orders of magnitude apart. If a model really is CPU-limited, `lz4` is nearly free; the compressor is a runtime knob, not a build-time decision.

**Prior art**: OpenWrt ships a maintained `zram-swap` package on targets with far less RAM than a BE-series router; Android has shipped zram by default since ~2013 on much weaker hardware; Chrome OS same. The trade was settled on slower silicon than anything this firmware supports.

## Level 1 — small backports, only if zswap+zsmalloc is adopted seriously

One real 4.19 hole, specific to picking zsmalloc as the zpool: **4.19's zsmalloc registers no `.shrink`** in its zpool driver, and `zpool_shrink()` is `driver->shrink ? … : -EINVAL`. So when the pool hits `max_pool_percent`, `zswap_shrink()` fails instantly and stores are rejected — zswap quietly **degrades to plain swap** from that point (it doesn't break; it just stops helping). zbud/z3fold do implement `.shrink`; this is the price of zsmalloc's density. Note this hole is invisible to zram — zram never asks the allocator to reclaim; you just shouldn't size a zram device optimistically.

Practical rule: **zswap+zsmalloc on 4.19 is strongly recommended only together with these patches; otherwise run z3fold/zbud** (which is what I did myself before backporting — z3fold first, zsmalloc after).

What I backported for it, all running on the live box:

| what | origin | depends on | how load-bearing |
|---|---|---|---|
| zswap LRU / pool shrinking | 6.5 `f999f38b4e6f`, hand-ported (1 file, +79/−5) | **nothing** — standalone (my port does single-entry reclaim in the store path’s full-handling; it went into my tree before the other two). **This is the one that lifts the z3fold-first rule** — with it, zsmalloc becomes a safe default | fixes the hole above. **Insurance, not a prerequisite** — my `pool_limit_hit` is still 0, it has never fired |
[... cut ...] ially to a zswap that old). Effort/benefit is upside-down there — if a 4.1 box needs compressed swap, just use zram.

Happy to point at individual commits if any of this is ever useful; none of it is a request. And to be clear about the original question: **Level 0 alone — zram as modules — already delivers most of the benefit above at zero cost to users who don't enable it.**

*(Investigation and measurements assisted by Claude under my direction; the box, the builds and the testing are mine.)*