gnuton/asuswrt-merlin.ng #586: 2.4 Channel Control

JordanCowper, 2024-05-09:
**Router Model Affected**
Models: (eg: DSL-AX82U/RT-AX82U/...) AX58U v2

**Firmware Version Affected**
3004.388.6_2-gnuton1 

**Is this bug present in upstream Merlin releases too?**
N/A - may have misunderstood this bit

**Describe the bug**
A clear and concise description of what the bug is.
Cannot manually choose channel

**To Reproduce**
Steps to reproduce the behavior:
1. Select manual channel on 2.4ghz
2. Apply
3. See no changes :(

**Expected behavior**
For UI and router to follow settings I've chosen

"CFG80211-ERROR) wl_dfs_cac_notify_status : 
May  9 22:08:27 kernel: In wl_dfs_cac_notify_status chanspec 0x1001 DFS state 0
May  9 22:08:27 acsd: eth2: selected channel spec: 0x1809 (7l)
May  9 22:08:27 acsd: eth2: Adjusted channel spec: 0x1809 (7l)
May  9 22:08:27 acsd: eth2: selected channel spec: 0x1809 (7l)
May  9 22:08:27 acsd: acs_csa_handle_request(359): eth2: err from dcs_handle_request:-22
May  9 22:08:27 acsd: acs_set_chspec: 0x1809 (7l) for reason ACS_INIT
May  9 22:08:27 acsd: acs_update_driver(440): acs update failed ret code: -22
May  9 22:08:27 acsd: acs_init_run(1249): eth2: update driver failed"
I'm not sure if the above is the cause :)

gnuton, 2024-05-10:
can you please upgrade to the latest beta and test it?
please try to reset to factory afterwards and test if it start to work.

gnuton, 2024-05-11:
Remember to factory reset and do not import settings or you will get the config messed up again.
Keep me posted.

gnuton, 2024-05-17:
any news here?

JordanCowper, 2024-05-17:
Ahh sorry no, I've been unable to update due to being quite busy I'll let you know as soon as possible :) I thought this was quite a popular router so surely if it was firmware we should've seen more have this problem

lord-dizzy, 2024-05-19:
Hello!
Indeed, I could confirm that both channels are affected by this issue (2.4 & 5 GHz). Neither the stable .6_2 nor the beta 7 is sticking with the selected channel (and extension). After selecting Apply, both boxes return to Auto.
And in Wireless log there are random channels selected, meaning that it's not just a display glitch, Auto is indeed effective.
AX58U v2 fully reset after each update.

JordanCowper, 2024-05-19:
I was about to flash tonight, thought it might have been a problem caused by me, but thanks for confirming everything else seems a-okay apart from this

Kuajmc, 2024-06-06:
Chiming in for more info:
Same issue here with a newly flashed RT-AX58U v2. However it seems to me that it is possible to choose 2,4 GHz channel via ash “nvram set wl0_chanspec=XX” followed by “nvram commit” and restart. It will still display Auto on WebUI, but it seems like the chosen channel stays after reboot. 

I haven’t tried with 5GHz as the wife needs Wi-Fi.

JordanCowper, 2024-06-06:
I appreciate that advice, I'll give this a whirl :)

gnuton, 2024-06-09:
Hi,
do you  get javascript erros in the browser coinsole when you try to change these values?

gnuton, 2024-06-09:
Checked and confirmed. no errors in the console.

gnuton, 2024-06-09:
This commit should fix this 9659a643676613eb46e13f158cb5a85fa841302b

JordanCowper, 2024-06-11:
I've updated and it seems to be following the settings. Great work Gnuton 👍

gnuton, 2024-06-11:
😎

closed 2024-07-04 by gnuton