gnuton/asuswrt-merlin.ng #588: 3004.388.7_1-gnuton0_beta1 Two questions.

qqw1231, 2024-05-10:
[image]
[image]

qqw1231, 2024-05-10:
[image]
[image]

qqw1231, 2024-05-10:
Figure 1 Login and password account login are not the same length.

Figure 2: The frame does not leave a gray low color.

qqw1231, 2024-05-11:
Figure 1 is the mobile phone login.

gnuton, 2024-05-11:
The login button is correct. This is the layout
[image]

gnuton, 2024-05-11:
Also the open NAT colors are intented to be like that.. 
The features comes from ROG and AFAIK the intention was to keep show it.

closed 2024-05-11 by gnuton