gnuton/asuswrt-merlin.ng #591: RT-AX82U refusing to update firmware

zugurudumba, 2024-05-16:
**Router Model Affected**
RT-AX82U

**Firmware Version Affected**
Currently on 3004.388.5-gnuton1, cannot update to 3004.388.6-gnuton1

**Is this bug present in upstream Merlin releases too?**
I do not know

**Describe the bug**
Currently on 3004.388.5-gnuton1. When trying to update to 3004.388.6-gnuton1 by uploading RT-AX82U_V2_3004_388.6_2-gnuton1_nand_squashfs.pkgtb, the UI goes through the update process, does not return any error and eventually the router restarts. Upon restart, the firmware version is still the same, 3004.388.5-gnuton1.

Tried from multiple browsers and PCs, same result.

**To Reproduce**
Prerequisite: RT-AX82U on firmware 3004.388.5-gnuton1.
1. Try to update to latest version, 3004.388.6-gnuton1.
2. Notice the update process seems to go through smoothly.
3. Notice the firmware does not actually update.

**Expected behavior**
Firmware should update.

**Screenshots**
[image]

zugurudumba, 2024-05-16:
A colleague trying to update to 3004.388.6-gnuton1 from stock firmware on the exact same model of router is encountering the same issue. I suspect there's something wrong with this package.

zugurudumba, 2024-05-16:
Updating both routers to 388.6_0-gnuton1 worked without issue.

gnuton, 2024-05-17:
@zugurudumba can we close this issue?

zugurudumba, 2024-05-17:
Yes, please. Thank you.

closed 2024-05-19 by gnuton