gnuton/asuswrt-merlin.ng #592: XT8 V1: Vesrions > RT-AX95Q_3004_388.5_0 cannot display AI Mesh nodes

colbsure, 2024-05-18:
**Router Model Affected**
Models: XT8 v1

**Firmware Version Affected**
3004.388.6_0 or newer

**Is this bug present in upstream Merlin releases too?**
Do not know.

**Describe the bug**
When installing version 6_0 or newer the AI Mesh page of the web interface is blank - looks like a javascript error is causing issues.

A fulll reset was tried with the broken version and did not help.

Problem is repeatable.  Merely upgrading causes the issue and downgrading solves it.  

Browser was Firefox but the issue is evident with Edge (a Chrome derivative).

**To Reproduce**
Steps to reproduce the behavior:
Upgrade firmware to 6_0, 6_2 or 7_1

**Expected behavior**
The AIMesh page shows as expected in the WebUI.

**Screenshots**
Version 6_0 is broken:
[image]

Version 5_0 is fine with no javascript error.
[image]

ajayramaswamy, 2024-05-19:
duplicate of #542

gnuton, 2024-05-20:
@ajayramaswamy is this something you have experienced too?
Do you get the same error in the JS console?

gnuton, 2024-05-20:
I just flashed the latest beta and the problem is not there.

[image]

closed 2024-05-20 by gnuton

gnuton, 2024-05-20:
I missed it was in access point mode. This happens only in that case. 
Closing as duplicate of #542