gnuton/asuswrt-merlin.ng #594: RT-AX82U , the error "nf_conntrack: expectation table full" appeared

linnvirgo, 2024-05-31:
**Router Model Affected**
Models: (RT-AX82U)

**Firmware Version Affected**
RT-AX82U_386.04_0-gnuton2_cferom_puresqubi.w

**Is this bug present in upstream Merlin releases too?**
I'M NOT SURE

**Describe the bug**
when i enabled the fullcone mode of the router, the error "nf_conntrack: expectation table full" appeared. the device upload network speed does not go up, disconnected from time to time, the number of network connections is only more than 1000,  the number of network connection is very low.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'system log'
2. Click on 'logs'
3. Scroll down to '....'  
4. See error

**Expected behavior**
As before , in fullcone mode, the number of connections reached more than 30000was normal, the upload speed did not drop to a few hundred kb, and the device did not disconnect from time to time.

gnuton, 2024-05-31:
This is a known issue whcih comes from upstream
https://www.snbforums.com/threads/nf_conntrack-expectation-table-full-and-other-log-oddities.55415/

linnvirgo, 2024-05-31:
> 这是一个来自上游的已知问题，这里应该有一个解决方法：https://www.snbforums.com/threads/nf_conntrack-expectation-table-full-and-other-log-oddities.55415/
what is the appropriate value for my router? Or do you try one at a time until there are no mistakes?

linnvirgo, 2024-05-31:
> This is a known issue whcih comes from upstream and there should be a workaround in here: https://www.snbforums.com/threads/nf_conntrack-expectation-table-full-and-other-log-oddities.55415/

After I set the value to 512, it only appeared twice in an hour. Can I set the value to 1024? Or have any better recommendations?

gnuton, 2024-06-01:
Hi,
I think it really depends on amount of connections you have. You can set it 1MB if you still have enough mem free