gnuton/asuswrt-merlin.ng #595: Support for RT-AX58U V2 - kill switch

aidancov1, 2024-05-31:
**Router Model Affected**
Models: (Asus AX 58U)

**Firmware Version Affected**
V2

**Is this bug present in upstream Merlin releases too?**
No, as it is not supported by Merlin

**Describe the bug**
I found your firmware yesterday and was pleased it supported the V2 of Ax58U. I installed and all seemed well but I am unable to get the kill switch to work. I have devices on WAN and OVPN and have used VPN Director to allocate different VPNs. When I disconnect the tunnel the devices start using the WAN side of the internet and don't get blocked. I've tried putting them on Strict but makes no difference. Any idea what could be wrong?

Thanks

Aidan

gnuton, 2024-06-10:
This feature should be backported in the next versions

gnuton, 2024-07-04:
This will be part of 388.8
1e4030ed7010c3758311cc3844f3bb7b10a4dc31

gnuton, 2024-07-05:
please test https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.8_0-gnuton0_alpha1

ajay-gh, 2024-07-06:
I understood that this adds more Asus routers to the list. I assumed that you took the original Merlin firmware code (e.g.  3004_388.7_0) and made minor changes in a limited number of places in order to do this. In other words, I assumed that the bulk of the code remains working, and stable. No need to continuous additional testing. If that assumption was correct, how come a feature like the VPN client's kill switch - which works fine in the Merlin firmware - does no longer work. 
Hope you don't mind explaining.