gnuton/asuswrt-merlin.ng #598: WireGuard not working?!

feiercrack, 2024-06-07:
**Router Model Affected**
RT-AX82U v1

**Firmware Version Affected**
3004.388.6_2-gnuton1
**Is this bug present in upstream Merlin releases too?**
IDK

**Describe the bug**
after downloading an wg config from surfshark, my ip adress is not masked with ip from surfshark, after enabling wg client in the vpn section
i still have the ip from my isp - vodafone germany 

[image]

[image]

it was working with original RT-AX82U_3.0.0.4_388_24232-gf74229b_puresqubi just before i upgrade it with your merlin

alfky12, 2024-06-08:
Add new rule in VPN Director tab.

feiercrack, 2024-06-08:
> Add new rule in VPN Director tab.

wow, awesome. that worked^^

ty!

closed 2024-06-08 by feiercrack

alfky12, 2024-06-08:
If you use IPv6, check whether there is a leak or not.